Atmosphere-NX/Atmosphere #2313: Error code 2123-0011 on Emunand when starting up a game

Mars0004, 2024-04-10:
## Bug Report

### What's the issue you encountered?

After upgrading Atmosphere to 1.7.0 and my HOS to 18.0.0, any game with online features(so all rest of games don't give me this error) on my emunand will give me an error code 2123-0011 after the nintendo logo disappear.  This happens when the switch is in my dock station.  I can just click then on the 'ok' button and the game will load as it used to be and the offline content I can play.

If my switch isn't docked I get also an error when starting same games, only then the error will be about no internet connection without any error code mentioned.  Also then if I click on 'cancel' on that error, the game will load like normal and I will be able to play the offline content.

I didn't had these error codes under Atmosphere 1.6.2 and HOS 17.0.1  I am only interested in the offline content of these games to play.

I do use MITM and exosphere to block all Nintendo dns servers as I also did before the upgrade.

### System Firmware Version

18.0.0</br>

### Environment?

EMUMMC (file based)
Atmosphere v 1.7.0
Hekate v 6.1.1

borntohonk, 2024-04-10:
> 2123-0011
> I do use MITM and exosphere to block all Nintendo dns servers as I also did before the upgrade.

https://github.com/Atmosphere-NX/Atmosphere/issues/2225#issuecomment-1831441277

either disable exosphere blanker or use this to patch out the fatal condition in NIM:

https://github.com/fruityloops1/nim-prodinfo-blank-fix/releases/tag/18.0.0

Mars0004, 2024-04-10:
> > 2123-0011
> > I do use MITM and exosphere to block all Nintendo dns servers as I also did before the upgrade.
> 
> [#2225 (comment)](https://github.com/Atmosphere-NX/Atmosphere/issues/2225#issuecomment-1831441277)
> 
> either disable exosphere blanker or use this to patch out the fatal condition in NIM:
> 
> https://github.com/fruityloops1/nim-prodinfo-blank-fix/releases/tag/18.0.0

Hi, I did try that nim prodinfo patch already last night without any luck.(with my wifi off, or must it be on?)  
The issue was still there.  And about the exosphere file, I am a bit worried.  Does it harm to disable the exosphere blanker in relation to online gaming on my stock?  Cause I do play official games online on my stock.

So I don't want my switch to get banned.  Or doesn't the disabling of the prodinfo blanker in exosphere get your switch banned?

my exosphere file looks like this at moment:
[exosphere]
debugmode=1
debugmode_user=0
disable_user_exception_handlers=0
enable_user_pmu_access=0
blank_prodinfo_sysmmc=0
blank_prodinfo_emummc=1
allow_writing_to_cal_sysmmc=0
log_port=0
log_baud_rate=115200
log_inverted=0

closed 2024-04-11 by Mars0004