Atmosphere-NX/Atmosphere #2315: Make a version for mig switch

triplean, 2024-04-14:
## Feature Request
Add a version of atmosphere for migswitch maybe some type of game. If you can do that you are going to save oled and lite players.

### What feature are you suggesting?
Get atmosphere in the migswitch
#### Overview:
Atmosphere in a cartridge, maybe some type of security to avoid bans

#### Smaller Details: 
You can do a id generator for every atmosphere rom so switch can't ban you. 

#### Nature of Request

### What component do you feel this would best fit within?
All

### Why would this feature be useful?
Because It doesn't matter which console you have, you can hack it with a migswitch.

Aura67, 2024-04-14:
The problem is not Atmosphere, I've been online with Atmosphere on the Nintendo server for over 3 years. What leads to a ban is piracy cheats and hacks online atmosphere is ok on its own even if you use it on the switch you don't need it on your mig card because it is used for piracy as we know piracy is not supported.

SciresM, 2024-04-14:
Piracy is not supported.

closed 2024-04-14 by SciresM

triplean, 2024-07-05:
"For legal purposes" it's not piracy

SciresM, 2024-07-05:
As I said, piracy is not supported.