Atmosphere-NX/Atmosphere #2318: Error al iniciar atmosphere

Marara98, 2024-04-19:
## Bug Report
Failed to identify FS
### What's the issue you encountered?
Cuando inicio el atmosphere aparece el logo y después manda un pantallaso con un texto que dice, 
"A fatal error occurred when running Fusée. 
Failed to identify FS!
Press POWER to reboot"
Al dar power se reinicia el atmosphere y hace lo mismo aparece el logo y salta el error, al sistema stock si entra y desde hakete probé a lanzar el fusée pero sigue marcando lo mismo

### How can the issue be reproduced?
N/A

### Additional context?
Sistema del emummc v17
Sistema stock en v18
SD no marca error al usarla en pc y desde usb en hekate además de probar una herramienta para sd en la switch
[image]

Aura67, 2024-04-19:
update Amospehre and hekate download the last release. You have probably installed firmware version 18.0.0 on your Emummc and are using atmospehre 1.6.2, which no longer works on firmware 18.0.0.

Marara98, 2024-04-19:
La versión de la consola si es 18 la emummc no, pero ya actualice de nuevo borrando la carpeta atmosphere por si era eso y también el hekate y no se soluciona, sigue saliendo el mismo fallo, puede ser fallo en la sd?
Que necesite empezar todo de cero o mejor esperar y ver si con una futura actualización se soluciona, que la memoria es de 256 y esta llena y sería bastante tiempo recuperar todo

antonwe, 2024-04-22:
this is not a support forum and we don't speak spanish here.

closed 2024-04-25 by Marara98