Atmosphere-NX/Atmosphere #2320: 0100000000000001 Error

naldo29, 2024-04-29:
### What's the issue you encountered?

Heres what happenend. After successfully modding my console and closing it up. I went into atmosphere and used it for a bit but then out of no where my console turned off, and now when i turn it on its now showing error 0100000000000001.

### How can the issue be reproduced?
By turning on the console

### Crash Report
[report_0000000006113d3f.zip](https://github.com/Atmosphere-NX/Atmosphere/files/15145086/report_0000000006113d3f.zip)

### System Firmware Version
18.0.0

### Environment?
Official Release
EmuMMC or SysNAND:
Sysnand

SciresM, 2024-04-29:
That report does not match the official pre-release's loader binary, and I think it's likely bogus.

Please try deleting your atmosphere directory and using the latest pre-release zip with absolutely zero add-ons or modifications?

naldo29, 2024-04-29:
[report_000000000810d6b5.zip](https://github.com/Atmosphere-NX/Atmosphere/files/15145150/report_000000000810d6b5.zip)
My bad for uploading the wrong file. I think this one should be the correct one

SciresM, 2024-04-29:
Do you have custom sysmodules? Please post a screenshot of atmosphere/contents and atmosphere/kips, if either dir is present.

naldo29, 2024-04-29:
i have kip_patches, but this is a brand new atmosphere install. all i did was download atmoshpere and hakete and thats it. thas all on the sd card. Nothing else

SciresM, 2024-04-29:
remove kip_patches, delete literally everything.

SciresM, 2024-04-29:
Please post screenshots of your unmodified atmosphere directory to confirm.

naldo29, 2024-04-29:
[image]
like this?

SciresM, 2024-04-29:
...okay, extract the latest pre-release zip. You do need to have atmosphere present, just no add-ons.

naldo29, 2024-04-29:
[image]
i deleted everything and extracted the latest release.

SciresM, 2024-04-29:
Okay. What happens when you try to boot the console?

naldo29, 2024-04-29:
Dont i need hakete to boot it up? i tried renaming the fusee.bin file as payload.bin so that the chip could use it but it does nothing.

SciresM, 2024-04-29:
If you are booting on a mariko unit, fusee must be chainloaded via hekate, yes. I did not assume you were booting on a mariko unit.

naldo29, 2024-04-29:
My apologies for not stating that. anyways the same issue did happen. but now its a different error id
[report_000000000e02e4f3.zip](https://github.com/Atmosphere-NX/Atmosphere/files/15145367/report_000000000e02e4f3.zip)

SciresM, 2024-04-29:
That report is yet another case of https://github.com/Atmosphere-NX/Atmosphere/issues/1256.

You have a corrupt PRODINFO or system settings save, you'll need to delete the system settings save and/or flash a non-corrupt PRODINFO. Anyway, this isn't an atmosphere bug, at this point.

SciresM, 2024-04-29:
I am guessing that your previous loader crash resulted from corruption-on-sd-card causing bad loader kip .text.

closed 2024-04-29 by SciresM