Atmosphere-NX/Atmosphere #2321: Error when booting (program 0100000000000005) Think its the number of 0

yoyoxnd, 2024-04-30:
## Bug Report

### What's the issue you encountered?

I was trying to download a mod for smash bros ultimate using Ultimate mod manager. I finished downloading the mod and on launch, i crashed. Ultimate mod manager may have changed something in the atmosphere folder, but not me.

### How can the issue be reproduced?

download a smash bros mod while trying to do as the tutorial say and (i guess) fail

### Crash Report

no crash report is created, but i have something in fatal error. Its not some simple text so i cant copy it.

This ,im pretty sure, is corrupted and undeletable.
[image]

### System Firmware Version

18.x.x (not sure and can't verify

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: official
  - version 17.0.0
- Do you have additional kips or sysmodules you're loading: I have, but they never caused a problem, even when updating
- Homebrew software installed: a few, but never had problem before Ultimate Mod Manager
- EmuMMC or SysNAND: EmuMMC partition based, but sysnand is yellow screening too

### Additional context?

Was not updating or something like that

I will still send the fatal error file, if you say to me where to send it

SciresM, 2024-04-30:
The file in fatal_errors is what I need. You can zip it and upload it.

But I'm almost entirely sure from your image that this is just "your SD card is corrupted". Almost certainly you will just need to wipe your SD card/re-clone fresh emummc/re-download and extract latest zips, etc.

yoyoxnd, 2024-04-30:
Can i just backup This sdcard then format it and pit back the backup with fully fresh atmosphere? I have my old sdcard which is like from a month or two ago if i can't. I could do a backup from this one.

yoyoxnd, 2024-04-30:
Can i just backup This sdcard then format it and pit back the backup with fully fresh atmosphere? I have my old sdcard which is like from a month or two ago if

closed 2024-04-30 by SciresM

yoyoxnd, 2024-04-30:
[report_0000000000a198b9.zip](https://github.com/Atmosphere-NX/Atmosphere/files/15169769/report_0000000000a198b9.zip)

yoyoxnd, 2024-04-30:
okay so i just have LOTS of corrupted file. Im taking what files are still good and will do without the others.