Atmosphere-NX/Atmosphere #2322: retroarch and melonds always crash when update firmware to 18.0.1

silentmissile, 2024-05-04:
my machine is OLED Japanese version

3 days ago, I update the firmware to 18.0.1, Atmosphere-1.7.0-prerelease

when I run retroarch or melonds, always crash with error code 2168-0002, program ID 010000000000100D

in Nintendo website, 18.0.1 updates Wi-fi issue, and error code 2168-0002 is network error code. I think they are related

E-R-Butch, 2024-05-04:
This is likely the 18.0.1 wifi bug. please delete all saved wifi profiles for now.

silentmissile, 2024-05-04:
> This is likely the 18.0.1 wifi bug. please delete all saved wifi profiles for now.

thank you for your proposal. 

I tried, but the problem is the same as before.

borntohonk, 2024-05-04:
> my machine is OLED Japanese version
>
> 3 days ago, I update the firmware to 18.0.1, Atmosphere-1.7.0-prerelease
>
> when I run retroarch or melonds, always crash with error code 2168-0002, program ID [...]

provide the fatal report / crash report, as you should have if you had used the issue template.

silentmissile, 2024-05-04:
[image]
in the picture, we can see the firmware version is 18.0.1, atmosphere version is 1.7.0 pre-release.

and the error code 2168-0002(0x4a8)
program: 010000000000100D

and the software retroarch is stable version
https://buildbot.libretro.com/stable/1.17.0/nintendo/switch/libnx/RetroArch.7z

before I update firmware, it ran smoothly. This is the reason I think the firmware update causes this problem. But we all know, we can't ask Nintendo to fix this issue :-(

then I report this issue to atmosphere

it's easy to reproduce this problem:
1) reboot the machine, select atmosphere in hekate menu
2) enter hbmenu
3) click RetroArch
4) the screen would become black for several seconds, then this error screen would appear

if you connect Wi-fi before step 3), the screen would be black for a little more time, maybe 1 minute, but finally the error screen would appear

Saeden, 2024-05-04:
I am also getting this error code when trying to run games in ppsspp. Do you want me to make my own issue ticket or can I supply the crash report here?

> This is likely the 18.0.1 wifi bug. please delete all saved wifi profiles for now.

I tried deleting all my wifi profiles but that didn't fix the issue. I am going to try to downgrade my firmware version to see if that fixes the issue.

silentmissile, 2024-05-04:
> Dirty... this isn't an official release.

the comment is 35d93a7c4, it's the newest release, it's 1.7.0-pre-release

silentmissile, 2024-05-05:
> dirty means additional changes were made and not commited. Where did you get this build from? Because you didn't download this from Atmosphere releases.

per your proposal, I download Atmosphere-1.7.0-pre-release again from

https://github.com/Atmosphere-NX/Atmosphere/releases/download/1.7.0-prerelease/atmosphere-1.7.0-prerelease-35d93a7c4+hbl-2.4.4+hbmenu-3.6.0.zip

The program still can't run. But the error hint is different. It is:

Firmware: 18.0.1 (Atmosphere 1.7.0-master-31ad4eec1)

all others are the same.

borntohonk, 2024-05-05:
> Firmware: 18.0.1 (Atmosphere 1.7.0-master-31ad4eec1)
> 

This and your image is not a fatal report or crash report, which you should have uploaded already had you filled out the issue template to begin with, and when asked, you instead posted an image.

Get it from your microSD card, in the appropriate folder i.e. sd://atmosphere/fatal_errors/

silentmissile, 2024-05-05:
> This and your image is not a fatal report or crash report, which you should have uploaded already had you filled out the issue template to begin with, and when asked, you instead posted an image. [...]

this is sd://atmosphere/crash_reports/01714908793_010000000000100d.log
[01714908793_010000000000100d.log](https://github.com/Atmosphere-NX/Atmosphere/files/15212716/01714908793_010000000000100d.log)

it's empty in sd://atmosphere/fatal_errors

silentmissile, 2024-05-05:
> https://docs.libretro.com/guides/install-libnx/#running-retroarch-using-title-takeover

thank you, it's resolved

closed 2024-05-05 by silentmissile