Atmosphere-NX/Atmosphere #2324: Panic

seriouslyunderappreciated, 2024-05-08:
### What's the issue you encountered?

Panic occurred while running Atmosphere.
Title ID: 010041544D530000
ERROR: DABRT (0x101)

### How can the issue be reproduced?

I wish I knew how to exactly reproduce the issue, but this crash DID happen under very similar circumstances twice across separate microSD cards, one of which is brand new, and the console was completely formatted (as in fresh sysnand) in between.

When I first opened a game and quickly went back to the home menu before it actually had a proper chance to boot.

### Crash Report

[report_d8dcf9dd.zip](https://github.com/Atmosphere-NX/Atmosphere/files/15245910/report_d8dcf9dd.zip)

### System Firmware Version

18.0.1|AMS 1.7.0|E

### Environment?

Hekate bootloader, latest official AMS (1.7.0 pre-release), no additional kips or sysmodules.

I have a folder for emuiibo with just amiibo files in it, but the actual emuiibo isn't installed, because I specifically wanted to rule out external causes of crash with a clean install of AMS on a new microSD and an EmuMMC made from a recently formatted SysNAND.

The EmuMMC is partition-based.

seriouslyunderappreciated, 2024-05-08:
I should point out that this doesn't happen every time or even often, but it did happen before on my previous emunand on a previous microSD card, which is why I took steps to rule out culprits by not adding anything to AMS in this particular case.

seriouslyunderappreciated, 2024-05-08:
Actually I neglected to mention I do have romFS mods installed for BOTW/TOTK.

borntohonk, 2024-05-08:
> When I first opened a game (before it had created the initial save files) and quickly went back to the home menu before it actually had a proper chance to boot (during the very first second of booting the game I pressed the home button and closed it with X) I got the fatal error.
> 
> I tried to delete the existing save data after this to recreate the steps that led to the panic, but I can't reliably make it happen. It happened before and again under similar circumstances

So basically, you take an IO hit from running emummc already, and you're trying to start a game, and before the game has started up and made its system save, you force the game into a paused mode by exiting to the home screen, and then you while having done that, close the game, and that forced the panic.

maybe don't do that? (the part where you open the game for barely a second, and then immediately exit to the home screen and closing the game)

seriouslyunderappreciated, 2024-05-08:
> So basically, you take an IO hit from running emummc already, and you're trying to start a game, and before the game has started up and made its system save, you force the game into a paused mode by [...]

I also had a freeze during normal initial loading before this. I'm not constantly doing what I described, that was just something I noticed happened twice :)

And crucially, I never had any of these problems on prior OFW/AMS.

masagrator, 2024-05-08:
> maybe don't do that?

I am doing this pretty often and never had issue with it that was related purely to Atmosphere. Only in one case this was causing panic on my Switch - if you have SaltyNX installed and you are booting Witcher 3. If you exit too fast, it causes fatal. This doesn't occur if SaltyNX is not running.

SciresM, 2024-07-15:
> I do have romFS mods installed for BOTW/TOTK, but they're awfully simple edits I made to some weapons so they don't break by adding an indestructible property to them. Very small files.

Yeah, this is caused by closing the game during a very specific window of romfs-building. With totk it's way easier to trigger, because totk has an obscene filecount and romfs building takes so long (relative to other games).

This is...already mitigated to the best of my ability, my best recommendation is just to not close the game during that window when using mods. Sorry!

closed 2024-07-15 by SciresM