Atmosphere-NX/Atmosphere #2325: Atmosphere crash 0100000000000013, 0x4A8 (2168-0002)

u549i, 2024-05-09:
### What's the issue you encountered?
Freeze and then atmosphere crash screen with 0100000000000013, 0x4A8 (2168-0002), followed by two crash reports 
second being for  0100000000000023, 0x25A0B (2011-0301). Happened two times so far after around 1-2 minutes into the game.

### How can the issue be reproduced?
Cannot reproduce it myself but it happened like 2-3 days ago first time and second time earlier today so i thought i might report. Happened after i booted the game and 1-2 minutes passed, game froze and then atmosphere crash 0100000000000013 screen appeared.

### Crash Report

[crash reports](https://gist.github.com/u549i/fdb04ffe529bbe52a3f7abf8eaedded1)

### System Firmware Version
18.0.1|AMS 1.7.0|E

### Environment?
Official release hekate 6.1.1, official release atmosphere 1.7.0 prerelease

I have OC Suite's fork (EOS) loader.kip with support of atmosphere 1.7.0
Also some "am" related stuff in atmosphere settings: 
`[am.debug]
continuous_recording_fps = u32!30
continuous_recording_video_bit_rate = u32!0x42D560
continuous_recording_key_frame_count = u32!15`
`[am.gpu]
gpu_scheduling_enabled = u8!0x0`
- Homebrew software installed: 
Sys-clk, Sys-tweak, SaltyNX, Edizon, Tesla ovlloader, NXThemes, ReverseNX-Tool, Status-monitor
Partition-based emummc

### Additional context?
I have a cartidrge in cartridge slot.

Masamune3210, 2024-05-10:
Try to reproduce the issue with a stock version of atmosphere, with no KIP's or extra modifications.

u549i, 2024-05-10:
I was using switch in scenario when changes to loader should be not relevant to cause this type of issue in my opinion, loader issues usually easily reproducable and have different error code. Can’t reproduce this issue anyway, with or without custom loader. Took 4 days of daily switch use for this to reappear. 
[image]

Masamune3210, 2024-05-10:
ANY modification can cause issues in entirely different places. Just because it doesn't seem like loader is involved doesn't mean it isn't involved. You can't just replace a crucial part of a system and not expect other parts to act differently.

u549i, 2024-05-13:
tried to reproduce the issue without custom loader (unsuccessfully) for two days and now i'm giving up and downgrading to 17.0.1 and will be waiting for new ams

closed 2024-05-13 by u549i

Masamune3210, 2024-05-15:
It sounds more like you need to watch out for a new version of the modification to loader to me. It's likely relying on patching something that is either in a different place than it is expecting or works differently than its expecting.