Atmosphere-NX/Atmosphere #2329: Fatal error  010000000000002b Std::abort <0xFFE>

Purpld, 2024-05-22:
## Bug Report
### What's the issue you encountered?

Keep getting error code labeled 
 title Id: 010000000000002b
Error:     Std::abort<0xFFE>

I didn't have a backup of my eMMC due to not big enough storage on my computer so I decided to just clean wipe and redo my switch mod I did it but then I get the exact same error message

### How can the issue be reproduced?
To be honest I don't know what has happened I clicked load on a game I went to the bathroom came back and I spit me this but I had literally just put my switch out of sleep mode then loaded it so I really don't know
If it helps the game was unturned

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

I can't find this out because I can't even load into atmosphere
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Official release
Atmosphère 1.7.0-prerelease
This github page
loading:
eMMC

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]
-

Purpld, 2024-05-22:
And if your wondering yes I looked at the other closed GitHub issue that was solved but I tried it all before I reset it didn't work

borntohonk, 2024-05-22:
> And if your wondering yes I looked at the other closed GitHub issue that was solved but I tried it all before I reset it didn't work

see https://github.com/Atmosphere-NX/Atmosphere/issues/2029#issuecomment-1414732456

closed 2024-06-02 by SciresM