Atmosphere-NX/Atmosphere #2330: Error Building Atmosphère with latest DevKitPro

impeeza, 2024-05-25:
Using the most recent version of DevKitPro on MSYS2's MinGW64 and latest packages:
- dkp-libs catnip 0.1.0-1
- dkp-libs deko3d 0.5.0-1
- dkp-libs devkita64-cmake 1.1.2-1
- dkp-libs devkita64-rules 1.1.1-1
- dkp-libs devkitarm-crtls 1.2.6-1
- dkp-libs devkitarm-rules 1.5.1-1
- dkp-libs dkp-cmake-common-utils 1.5.2-1
- dkp-libs dkp-toolchain-vars 1.0.3-2
- dkp-libs libnx 4.6.0-1
- dkp-libs switch-cmake 1.5.1-1
- dkp-libs switch-examples 20231028-1
- dkp-libs switch-glm 0.9.9.7-2
- dkp-libs switch-libdrm_nouveau 1.0.1-2
- dkp-libs switch-libjpeg-turbo 2.1.2-2
- dkp-libs switch-mesa 20.1.0-4
- dkp-libs switch-pkg-config 0.28-4
- dkp-libs switch-sdl2 2.28.5-4
- dkp-windows devkit-env 1.0.1-2
- dkp-windows devkitA64 r25.1-1
- dkp-windows devkitA64-gdb 14.1-1
- dkp-windows devkitARM r64-2
- dkp-windows devkitpro-keyring 20180316-1
- dkp-windows general-tools 1.4.4-1
- dkp-windows hactool 1.4.0-3
- dkp-windows switch-tools 1.12.0-1
- dkp-windows uam 1.1.0-2

Even with latest libnx build from https://github.com/switchbrew/libnx applying the PR 641 the build ends with lots and lost of errors on the «fusee_sdram.cpp» file, errors like:

```
/Atmosphere/fusee/program/source/sdram/fusee_sdram.cpp:1012:58: error: right shift count >= width of type [-Werror=shift-count-overflow]
 1012 |                             cur_reg_value |= ((src_value >> (SrcLow - DstLow)) & Mask);        \
```

a screenshot for reference:

[image]

The error is the same using UBUNTU and same set of packages

There is any package or step I am Missing on the building?

Thanks a lot for all your hard work.

SciresM, 2024-05-25:
I am waiting for devkitA64 to be updated. The fact that devkitARM is updated but not devkitA64 does mean that atmosphere can't be built with latest packages for now, yes.

impeeza, 2024-05-26:
Thank you so much forthe quick answer.

And thank you for all your great work and effort. 

If you don't mind, I will leave the issue open just for reference until all packages get updated

impeeza, 2024-05-30:
Hello there,  Thanks for all your work,  On my setup:

On a completely new installed and fresh MSYS2 environment using MSYS:

INSTALL PREREQUISITE PACKAGES
```
Pacman -Syuu --noconfirm --needed zip git make lz4 libnx devkitA64 devkitARM devkitarm-rules hactool switch-dev dkp-toolchain-vars switch-tools switch-mesa switch-libdrm_nouveau switch-sdl2 switch-libjpeg-turbo switch-glm gcc python python-pip
pip install lz4 pycryptodome
```

UPGRADE LIBNX TO LATEST CODE and APPLYING THE PR # 641
```
cd ~
rm -rf ~/libnx
git clone --recursive https://github.com/switchbrew/libnx.git
cd ~/libnx
git fetch origin pull/641/head
git checkout -b pullrequest FETCH_HEAD
make all -j$(nproc)
make install -j$(nproc)
```

#putting my PROD.KEYS AND PUT THEM ON THE `~/.switch` FOLDER as `prod.keys` file
```
mkdir ~/.switch
cp "MYPROD.KEYS" ~/.switch/prod.keys
```

Building Atmosphère code using the Branch gcc-14
```
cd ~
git clone --recursive https://github.com/Atmosphere-NX/Atmosphere
cd ~/Atmosphere
git checkout gcc-14
make
```

Sadly didn't worked for me, I got these errors:

[image]

[image]

```makefile
built ... loader_stub.bin
Checked loader stub.
Built fusee.bin...
result_get_name.cpp
In file included from T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results.hpp:23,
                 from T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours.hpp:28,
                 from T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/source/result/result_get_name.cpp:20:
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:93:60: error: expected identifier before 'this'
   93 |                 constexpr ALWAYS_INLINE BaseType GetModule(this auto const &self) { return ResultTraits::GetModuleFromValue(self.GetValue()); }
      |                                                            ^~~~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:93:60: error: expected ',' or '...' before 'this'
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:94:65: error: expected identifier before 'this'
   94 |                 constexpr ALWAYS_INLINE BaseType GetDescription(this auto const &self) { return ResultTraits::GetDescriptionFromValue(self.GetValue()); }
      |                                                                 ^~~~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:94:65: error: expected ',' or '...' before 'this'
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp: In member function 'constexpr ams::result::impl::ResultBase::BaseType ams::result::impl::ResultBase:
GetModule(int)':
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:93:125: error: 'self' was not declared in this scope
   93 |                 constexpr ALWAYS_INLINE BaseType GetModule(this auto const &self) { return ResultTraits::GetModuleFromValue(self.GetValue()); }
      |
                                ^~~~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp: In member function 'constexpr ams::result::impl::ResultBase::BaseType ams::result::impl::ResultBase:
GetDescription(int)':
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:94:135: error: 'self' was not declared in this scope
   94 |                 constexpr ALWAYS_INLINE BaseType GetDescription(this auto const &self) { return ResultTraits::GetDescriptionFromValue(self.GetValue()); }
      |
[... 3 lines cut ...]
  122 |             constexpr ALWAYS_INLINE typename Base::BaseType GetModule() const { return Base::GetModule(); }
      |                                                                                        ~~~~~~~~~~~~~~~^~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:93:50: note: candidate: 'constexpr ams::result::impl::ResultBase::BaseType ams::result::impl::ResultBase::GetModule(int)'
   93 |                 constexpr ALWAYS_INLINE BaseType GetModule(this auto const &self) { return ResultTraits::GetModuleFromValue(self.GetValue()); }
      |                                                  ^~~~~~~~~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:93:50: note:   candidate expects 1 argument, 0 provided
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp: In member function 'constexpr ams::result::impl::ResultBase::BaseType ams::Result::GetDescription() const':
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:123:113: error: no matching function for call to 'ams::Result::GetDescription() const'
  123 |             constexpr ALWAYS_INLINE typename Base::BaseType GetDescription() const { return
Base::GetDescription(); }
      |
~~~~~~~~~~~~~~~~~~~~^~
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:94:50: note: candidate: 'constexpr ams::result::impl::ResultBase::BaseType ams::result::impl::ResultBase::GetDescription(int)'
   94 |                 constexpr ALWAYS_INLINE BaseType GetDescription(this auto const &self) { return ResultTraits::GetDescriptionFromValue(self.GetValue()); }
[... cut ...]
T:/msys64/home/Impeeza/Atmosphere/libraries/libvapours/include/vapours/results/results_common.hpp:94:50: note:   candidate expects 1 argument, 0 provided
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:16: result_get_name.o] Error 1
make[2]: *** [/home/Impeeza/Atmosphere/libraries/config//../libstratosphere/libstratosphere.mk:111: all] Error 2
make[1]: *** [/home/Impeeza/Atmosphere/atmosphere.mk:152: libstratosphere] Error 2
make[1]: Leaving directory '/home/Impeeza/Atmosphere'
make: *** [Makefile:39: nx_release] Error 2

``` 

Again Thanks a lot for all your hard work.

SciresM, 2024-05-30:
@impeeza that branch requires gcc 14 devkitA64, which has not been released. I have not closed this issue or merged that code for that reason.

impeeza, 2024-06-02:
With the release of a new devkitA64 I was able to build latest code, even after merging the «gcc-14» branch on the main one and the code build flawless!

Thanks a lot for your help, 

Closing the issue.

commit a0793063 (2024-05-26): ams: fix compilation with gcc 14 (closes #2330)
commit f35c9481 (2024-06-02): ams: fix compilation with gcc 14 (closes #2330)
closed 2024-06-02 by impeeza