Atmosphere-NX/Atmosphere #2338: switch oled mariko not loading game cartridge need fixing

BladeTechBlade, 2024-06-09:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

borntohonk, 2024-06-09:
> I have got a modchip switch oled but one of my switch games does not load with hats pack need to check other packs to see if it is not a problem.
> 
> Iam on the latest switch 18.0.1 and the latest hekate build. A software for the switch should be made to identify if the problem is the game or the switch itself.

HATS is a piracy package, piracy is not supported.

borntohonk, 2024-06-10:
> It has got my game to work thanks from the people that made the cfw and I do not like anyone saying anything bad that helps people out.

The "people who made the cfw" are the contributors of this repository that you've submitted an issue to.

H.A.T.S. is an acronym for Hekate, Atmosphere, Tinfoil and Signature patches.

HATS is not a cfw, it's a piracy package, which is not supported. This github iissue does not belong here.

BladeTechBlade, 2024-06-10:
> > It has got my game to work thanks from the people that made the cfw and I do not like anyone saying anything bad that helps people out.
> 
> The "people who made the cfw" are the contributors of this repository that you've submitted an issue to.
> 
> H.A.T.S. is an acronym for Hekate, Atmosphere, Tinfoil and Signature patches.
> 
> HATS is not a cfw, it's a piracy package, which is not supported. This github issue does not belong here.

OK thanks

closed 2024-07-15 by SciresM