Atmosphere-NX/Atmosphere #2341: Update causes Black Screen (18.0.1 -> 18.1.0)

burnedpopcorn, 2024-06-11:
## Bug Report

### What's the issue you encountered?

After accidently updating from v18.0.1 to 18.1.0, Atmosphere refuses to boot, and becomes stuck on a Black Screen after the Atmosphere boot screen

I even changed in a Fresh SD Card with only the Latest Atmosphere, Hekate, and Sys-Patches and SigPatches, and this issue was still present

Also tried my main SD card without updating sig or sys-patches and re-installing everything on my main one
> (Both are FAT32)

### How can the issue be reproduced?

Step 1: Update to the Latest Official Firmware (18.1.0)
Step 2: Launch fusee.bin within hekate
Step 3: Feel extreme dread knowing nothing works

### Crash Report

I could not find a crash report in either SD Cards

### System Firmware Version

18.0.1 -> 18.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
> Hekate 6.1.1
- Official release or unofficial build: 
> Atmosphere 1.7.0
- Do you have additional kips or sysmodules you're loading:
> Got sigpatches and sys-patches from this website (didn't install anything else) :
> https://sigmapatches.su/
weird ass website
- Homebrew software installed:
> Way to much to put here, but all homebrew stayed within my main SD card, and not internal memory
> Also didn't install any new homebrew before updating (haven't installed anything new since last fusee.bin launch)

FourHawk, 2024-06-11:
This is expected, new HOS updates tend to ether break Atmosphere, hekate, or most commonly, both, the simple fix is to wait for an update on both sides

borntohonk, 2024-06-11:
> * Do you have additional kips or sysmodules you're loading:
> 
> 
> > Got sigpatches and sys-patches from this website (didn't install anything else) :

Piracy is not supported.

closed 2024-06-11 by SciresM