Atmosphere-NX/Atmosphere #2342: Panic occured while running Atmosphere.

EmmettVA, 2024-06-11:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

### What's the issue you encountered?

Updated Atmosphere and now there's this error every time it starts:
Panic occured while running Atmosphere.
Title ID: 010000000000D609
Error: std::abort (0xFFE)</br>

### How can the issue be reproduced?

Download the latest hekate and atmosphere from their respective github pages and move the folders to the sd.
Inject hekate bin, select sysnand and you will get it.</br>

### System Firmware Version

18.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unoffical build?
  - Official release version 1.7.1
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: aio switch updater, amiigo mod, appstore, deepsea toolbox, edizon, goldleaf, jksv, nxdumptool, nx-shell, super modyssey loader, switchident gui, and sys-clk manager
- EmuMMC or SysNAND: sysnand

### Additional context?

- Additional info about your environment: none

borntohonk, 2024-06-12:
> ### How can the issue be reproduced?
> 
> Download the latest hekate and atmosphere from their respective github pages and move the folders to the sd. Inject hekate bin, select sysnand and you will get it.

Shouldn't you file this issue in https://github.com/CTCaer/hekate/issues then? The "sysnand" option uses hekates package3 method rather than fusee.

> * Do you have additional kips or sysmodules you're loading: no
> 
> *  sys-clk manager
Those are contradicting statements.

EmmettVA, 2024-06-12:
i will

closed 2024-06-12 by EmmettVA