Atmosphere-NX/Atmosphere #2343: switch lite screen blurred

sir1st, 2024-06-12:
## Bug Report

### What's the issue you encountered?

switch lite screen blurred when update the ams from 1.7.0 to 1.7.1
[image]

I use the latest module         
 "atmosphere",
"hekate",
"deepseaassets",
"hbappstore"
to combine a new cfw with ofw version 18.1.0

either the ofw and cfw system crashes.
but when i downgrade the cfw to 1.7.0, i can go to the ofw successfully, while cfw gets crashes.

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

no logs in screen, just blurred and stop responding.

### System Firmware Version

18.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - [ Official release version 1.7.1 ]
- Do you have additional kips or sysmodules you're loading: no

### Additional context?

borntohonk, 2024-06-12:
> ## Bug Report
> ### What's the issue you encountered?
> 
> switch lite screen blurred when update the ams from 1.7.0 to 1.7.1 ![cb3655324a3271589de6722d9786c29](https://private-user-images.githubusercontent.com/23466852/338830225-2a1c889d-980c-4f71-b45c-3a35ef5942e6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTgxNzM3MTEsIm5iZiI6MTcxODE3MzQxMSwicGF0aCI6Ii8yMzQ2Njg1Mi8zMzg4MzAyMjUtMmExYzg4OWQtOTgwYy00ZjcxLWI0NWMtM2EzNWVmNTk0MmU2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjEyVDA2MjMzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1MjdhMzI1OTQxZjQ2OWNlMzEwNmUwZTQ4ZDlkMzgxYmFmYTBiNmQ1ODBhMzZlNjZmZjY0NTcwNmE4ZTQxNDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.BnTixRvkWhllsKfBWXj-uiZbYjdPPkO9eva1V1GNDXc)
> 
> I use the latest module "atmosphere", "hekate", "deepseaassets", "hbappstore" to combine a new cfw with ofw version 18.1.0
> 
> either the ofw and cfw system crashes. but when i downgrade the cfw to 1.7.0, i can go to the ofw successfully, while cfw gets crashes.
> ### How can the issue be reproduced?
> 
> [ * ] [ Include a detailed step by step process for recreating your issue. ]
> ### Crash Report
> 
> no logs in screen, just blurred and stop responding.
> ### System Firmware Version
> 
> 18.1.0
> ### Environment?
> 
>     * What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
> 
>     * Official release or unofficial build:
>       
>       * [ Official release version 1.7.1 ]
> 
>     * Do you have additional kips or sysmodules you're loading: no
> 
> 
> ### Additional context?

I'm pretty sure you should redirect your support queries here, as your picture shows their bootlogo: https://github.com/Team-Neptune/DeepSea-Assets/issues

SciresM, 2024-06-12:
Closing this. This is for ams bugs, not for support or other non-ams issues. If the issue occurs with plain atmosphere with nothing else, re-opening could be considered, but this is almost certainly not an ams issue.

closed 2024-06-12 by SciresM