Atmosphere-NX/Atmosphere #2345: Atmosphere 1.7.1 Panic error after CFW Upgrade from 18.0.1 to 18.1.0

SnowySorcerer, 2024-06-15:
## Bug Report

"N/A"

### What's the issue you encountered?

Panic occurred (Fatal Error) while running Atmosphere 1.7.1.

Title ID: 010000000000BD00
Error: std::abort (0xFFE)

### How can the issue be reproduced?

Launching CFW - emuMMC via hekate
Atmosphere 1.7.1 following CFW Upgrade from 18.0.1 to 18.1.0

### Crash Report

[report_013f6988b.zip](https://github.com/user-attachments/files/15847701/report_013f6988b.zip)

### System Firmware Version

18.1.0

### Environment?

18.1.0
hekate_ctcaer_6.2.0_Nyx_1.6.2_v3
Atmosphere 1.7.1
- EmuMMC partition-based

### Additional context?

Hi guys, I updated to Atmosphere 1.7.1 (and hekate_ctcaer_6.2.0_Nyx_1.6.2_v3) as it showed support for 18.1.0. I updated the System firmware and then updated CFW to 18.1.0 via daybreak.

However upgrading to 18.1.0 was a big mistake as I now get a Atmosphere panic attack. Can you please advise? I have attached the fatal error report.

masagrator, 2024-06-15:
You didn't update MissionControl. Either update it or remove folder "010000000000BD00" from atmosphere/contents

SnowySorcerer, 2024-06-15:
Thank you so much.. I wasn't aware that I needed to update MissionControl but I have and it resolved the issue! Thanks a lot for the quick Solution. #Legends

closed 2024-07-15 by SciresM

SagastumeAaron, 2024-08-02:
hola @SnowySorcerer y como es que lo actualizaste?? es decir en donde encontraste los archivos actualizados de MissionControl, ya que estoy con los mismos problemas