Atmosphere-NX/Atmosphere #2346: Built in 8gb ram mode

caden3ds, 2024-06-20:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
A toggle for enabling a 8gb ram mode so you don't have to compile Atmosphere yourself

#### Smaller Details: 
I'm referring to the MemoryMode and MemorySize properties seen in this video https://www.youtube.com/watch?v=J4oXmQGZgzU at 22:52.

#### Nature of Request:  
Addition
### What component do you feel this would best fit within?
Not sure
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
Convenience and not having to rebuild atmosphere every time there is a new update

gh0sti, 2024-07-06:
This feature request is for specific hardware mod. There is probably very few users who can mod their device to add more ram.

SciresM, 2024-07-15:
I think I would prefer that people just compile with the relevant changes...this shouldn't be a big deal to the kind of people willing to manually install bigger RAM, and it would be confusing/problematic for normal users to change.

closed 2024-07-15 by SciresM