Atmosphere-NX/Atmosphere #2351: Updated to atmosphere-1.7.1-master-39c201e37+hbl-2.4.4+hbmenu-3.6.0 then updated firmware from 17.0.1 to 18.0.1 then updates emuMMC via https://nh-server.github.io/switch-guide/extras/updating/ guide

jimmiejoe, 2024-06-25:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>
Followed the guide "https://nh-server.github.io/switch-guide/extras/updating/" to update prior to updating the Switch firmware. Updated to atmosphere-1.7.1-master-39c201e37+hbl-2.4.4+hbmenu-3.6.0
Update Hekate
Then updated Switch firmware from 17.0.1 to 18.0.1
Then updated emuMMC via TegraExplorer.bin firmware backup and GoldLeaf.
Upon restart got an error, restarted again and got a panic error.

Rebuild the emuMMC by following the "https://nh-server.github.io/switch-guide/user_guide"as though starting CFW from scratch.
wrote the new emuMMC partition over the existing emuMMC partition, but this time the switch was on 18.0.1, instead of 17.0.1.
Launched Atmosphere and again got panic error:
Tried to launch Atmospere from both Hekate (Atmosphere FSSO emuMMC) and from fusee.bin payload

Panic occurred while running Atmosphere

Title ID: 010000000000BD00
Error:     std: :abort  (0xFFE)
Report saved to atmosphere/fatal_error/report_0a1bc77a.bin

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

18.0.1</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate and fusee
- Official release or unofficial build: official release
  - [ Official release version x.x.x (or) unofficial build ]1.7.1
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]
Partition emuMMC
### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

jimmiejoe, 2024-06-25:
[report_0a1bc77a.zip](https://github.com/user-attachments/files/15964849/report_0a1bc77a.zip)

borntohonk, 2024-06-25:
> [report_0a1bc77a.zip](https://github.com/user-attachments/files/15964849/report_0a1bc77a.zip)

delete folder 010000000000BD00 in atmosphere/contents/ or update mission-control ( https://github.com/ndeadly/MissionControl/releases )

jimmiejoe, 2024-06-25:
Updating Mission Control did not fix the issue, get the same error. I saw that in the previous Panic error.

borntohonk, 2024-06-25:
> Updating Mission Control did not fix the issue, get the same error. I saw that in the previous Panic error.

010000000000BD00 is missioncontrol related and is not an official titleid belonging to any atmosphere components nor nintendo, it is entirely related to the custom homebrew missioncontrol.
Delete the folder or update it properly. You claim to have updated it, and it doesn't work for you, which leaves you with the option of deleting it.

SciresM, 2024-06-25:
The issue reported is an abort in Mission Control; update it. This isn't an atmosphere issue.

closed 2024-06-25 by SciresM

jimmiejoe, 2024-06-25:
I do not know what Mission Control is and obviously do not use it.
I have deleted the folder and that appears to have corrected the issue.

not sure why the conflict only surfaced after I updated the emcMMC to 18.0.1.

masagrator, 2024-06-25:
It doesn't change the fact that you had MissionControl. You should get more aware of what you have installed on sdcard if you didn't know you had it.