Atmosphere-NX/Atmosphere #2353: 1) piracy is not supported

sousa-jeferson, 2024-06-26:
1) piracy is not supported
2) that's not how any of this works

_Originally posted by @SciresM in https://github.com/Atmosphere-NX/Atmosphere/issues/2352#issuecomment-2190585239_
            
[Atmossphere version for mig switch]

Feature Request

Add a version of atmosphere to migswitch.
Because I currently have a 1TB SD, it has
98% with Games installed, highlighting that my games are original.
And I have another 256GB SD, I wanted to install 1 Linux, however, I always have to remove the SD, there was a day when I kept removing it, it didn't recognize the SD, I thought I had burned the 1TB SD.

I work in IT, and sometimes I need support restarting some services, so I don't have to leave Nintendo's side.

What feature are you suggesting?

Possibility of atmosphere in the migswitch.

Overview:

Atmosphere on a Migswitch.

Minor details:

To make it easier for users of different operating systems not to keep changing the SD all the time, with the possibility of burning the SD.

Nature of the Request
Which component do you think this would fit best into?

All

Why would this feature be useful?

Because no matter what operating system the user has, they can be flexible using the migswitch and SD.

SciresM, 2024-06-26:
Again, this is not how this works.

closed 2024-06-26 by SciresM

sousa-jeferson, 2024-06-26:
Poderia explicar melhor por favor?