Atmosphere-NX/Atmosphere #2354: Fusee Fatal Error: Unknown EmcDvfsTimingTableIndex: -1

SanMi-El-Gordo, 2024-06-29:
##Bug report

### What is the problem you encountered?

Launching atmosphere with fusee and with fss produces the error "Fusee fatal error: Unknown EmcDvfsTimingTableIndex: -1"
Mariko console acquired 06/15/24, memory hynix atmosphere 1.7.1 pure Hekate 6.2.0 pure

### How can the problem be reproduced?
N/A
### Accident report
N/A

### System firmware version

18.1.0

### Atmosphere?

- What bootloader (fusèe, hekate, etc.) was Atmosphère launched with: fusee and hekate
- Official release or unofficial build: Official release version 1.7.1
- Do you have additional system kips or modules that you are loading? No
- Homemade software installed: No
- EmuMMC or SysNAND: EmuMMC based on partitions.

### Additional context?

N/A

SciresM, 2024-06-29:
What does hekate say your device's dram id is (in hardware info)? Asking about the number.

Maybe we're missing newer timing tables, somehow.

SanMi-El-Gordo, 2024-06-30:
29- Hynix H54G46CYRBX267 4GB

hexkyz, 2024-07-10:
The MTC tables for the new 16.0.0+ DRAM ids were missing. This should now be fixed in the latest commit.

closed 2024-07-15 by SciresM