Atmosphere-NX/Atmosphere #2355: HBMenu applet mode freezes atmosphere

Gabriele73, 2024-06-29:
## Bug Report
Launching hbmenu in applet mode (album) while a game is in the background causes Atmosphere to become unresponsive.
[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
Launching hbmenu in applet mode (album) while a game is in the background causes Atmosphere to become unresponsive, forcing a force reboot by holding power button (system reboot menu does not respond either) 
[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
1) Launch a game ( any is fine)
2) Return to homescreen
3) Open hbmenu by clicking on album
4) Freezes in a grey screen making system unresponsive
[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>
N/A
### System Firmware Version

17.0.1 | AMS 1.7.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: 
  - Official release 17.0.1
- Do you have additional kips or sysmodules you're loading:
- SaltyNX, MissionControl, SysDVR (I've tried using no sysmodules, same issue)
- Homebrew software installed: Daybreak, Goldleaf, Edizon, appstore, checkpoint
- EmuMMC or SysNAND:
  - EmuMMC partition based

SciresM, 2024-06-29:
I am almost completely sure this is a user configuration issue...I opened hbmenu with a game open totally fine just now.

can you rename the atmosphere dir temporarily + extract totally unmodified release zip -> post a screenshot of your atmosphere/contents folder showing absolutely nothing inside, and share what game you're testing with? Also worth checking if hbmenu/hbl.nsp got corrupted.

Gabriele73, 2024-06-29:
Is there an accurate way to verify if the hbl got corrupted? By the way, proceeding to follow the other instructions too.

SciresM, 2024-06-29:
Well, you could sha256sum the file on sd vs the file in the zip?

urherenow, 2024-07-03:
I 've never had this issue either. I jump into hbmenu often while playing games to play around with Edizon.

AK478BB, 2024-07-05:
I think too much sysmodule plugins running in background that cause the console stun,  please holding the power button restart the console,   then you should turn off them via deepseatoolbox.nro in hbmenu or diable it startup by commonproblemresolver.bin in Hekate

SciresM, 2024-07-15:
Closing due to inactivity.

closed 2024-07-15 by SciresM