Atmosphere-NX/Atmosphere #2356: ERROR: KRNL (0XF00)

strapppy, 2024-06-30:
## Bug Report

Trying to enter Atmosphere, CFW, OFW or stock causes the switch screen to go dark. Can still go into recovery mode and access Hekate.

### What's the issue you encountered?

I had not touched my switch for a month and wanted to update atmosphere to get newest version. Downloaded the Switch CFW 18.1.0.rar from BetterGaming and replaced my files with those in the pack. Issues starting occurring after that.

### How can the issue be reproduced?

I deleted all of my files in the SD card, and replaced it with BetterGaming 18.1.0.rar, same issue
Also removed all files again and tried to just install atmosphere and hekate from github on my SD card. 
Got same error, could not boot into stock or CFW or Atmosphere. 

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
https://gist.github.com/strapppy/a8c5abc612e0c8d9e715f41a387fd548

### System Firmware Version

17.0.0</br>
Tried updating to 18.1.0 and errors
### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate 
- Official release or unofficial build: 
  - Official release version 18.1.0
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: N/A
- EmuMMC or SysNAND:
  - Partition EmuMMC

### Additional context?

- Additional info about your environment:
- It's a fatal error with ID 524850534F53454D

borntohonk, 2024-06-30:
> I deleted all of my files in the SD card, and replaced it with BetterGaming 18.1.0.rar, same issue
> Also removed all files again and tried to just install atmosphere and hekate from github on my SD card.
> Got same error, could not boot into stock or CFW or Atmosphere.

Shouldn't you be seeking support from his youtube page or patreon? Since the package you reference is something he distributes to his patreon subscribers...? He also seems to be stupid enough to give his paying patreon subscribers access to pirated titles...

strapppy, 2024-06-30:
> Shouldn't you be seeking support from his youtube page or patreon? Since the package you reference is something he distributes to his patreon subscribers...? He also seems to be stupid enough to [...]

I have tried but there hasn't been much discussed or anyone available to help me on his servers. Fortunately I have discovered that KRNL errors are typically hardware errors in RAM or CPU

SciresM, 2024-06-30:
Kernel issues these days are actually mostly people booting using new atmosphere but outdated hekate. Anyway, support not provided for issues using unofficial packs.

closed 2024-06-30 by SciresM

Madara1-lang, 2025-01-14:
Как решить проблему то? Выскочила точно такая же.