Atmosphere-NX/Atmosphere #2357: haze USB compatibility issue

urherenow, 2024-07-03:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

USB connection works immediately with a USBC -> USB A cable, but is not recognized at all using a USB C -> USB C cable. This same cable works in hekate/NYX, and it also works in haze, if I plug it into a C<->A adapter.</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

1) Open haze </br>
2) Plug in USB C -> USB C cable </br>
3) Nothing happens </br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

N/A </br>

### System Firmware Version

18.1.0</br>

[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

hekate, Official release of both. </br>

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

N/A </br>

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2024-07-15:
I don't actually know enough about the usb APIs to debug this; I wouldn't expect that type-of-cable would be relevant to the usage of usb:ds. Do other homebrew tools making use of usb:ds work?

Also, worth noting that haze's author(/nominal maintainer) is no longer working on Switch-related projects, so haze stuff is harder than other modules to get sorted.

SciresM, 2024-07-16:
@urherenow Yeah, haze is open source. USB init/config is done here: https://github.com/Atmosphere-NX/Atmosphere/blob/master/troposphere/haze/source/usb_session.cpp#L72

SciresM, 2024-07-16:
Whoops, didn't mean to close this.

urherenow, 2024-07-18:
So from CTCaer "nyx forces device mode by not setting the spdsrc bit which allows power to be enabled for otg. your pc probably favors device mode. there are several laptops/dekstops that have that bs behavior. it's possible to do a role swap if I remember with hos ipc"

So I'm not sure if this is something I can correct on my pc and laptop, and I'm not sure if it's related to hos behavior or haze's behavior.

urherenow, 2024-07-18:
This could go either way, I guess. On one hand, there might be a way to force the correct mode as CTCaer said, and on the other hand, this is ultimately a result of my PC's behavior. It's up to you to decide if the juice is worth the squeeze, but keep in mind that it's several brands that do this, and not just mine (my PC is AMD-based, and my laptop is Inel...). 

If you might be interested in looking into this in the future, please feel free to re-open, but I'll close the issue since it's way over my head and not necessarily a bug.

closed 2024-07-18 by urherenow

SciresM, 2024-07-18:
@urherenow i don't think any such IPC exists? USB provides usb:ds for device mode and usb:hs for host mode, and we use usb:ds and configure the same way Nintendo does. IPC reference here: https://switchbrew.org/wiki/USB_services -- I don't see anything that does what you say.

If you have a specific change to make that would help I would do so, but yeah not actually sure what you want is even possible.