Atmosphere-NX/Atmosphere #2361: Region override does not work on Fitness Boxing Feat. Hatsune Miku 1.01

fvig2001, 2024-07-14:
## Bug Report

### What's the issue you encountered?
Region override does not work on latest version of the game. 

Added config.ini for the game:
[override_config] 
override_language = ja 
override_region = jpn
to contents/010026801E0AC000
and ran the game using latest version.

Game should be in Japanese (multi-language game depended on language setting). Actual game is still using current language set on Switch (English).

On 1.0 of game, it was running in Japanese via override_config on an English system.

#Steps
1. Set Switch to use non Japanese language in Settings and restart if needed
2. Create config.ini with values:
[override_config] 
override_language = ja 
override_region = jpn
3. Save to sd:/atmosphere/contents/010026801E0AC000 (ID of the game)
4. Run game using version 1.01
Expected is game runs in Japanese language

I tested using Ryujinx and changing language on settings has correct behavior for latest version.

### Crash Report
N/A

### System Firmware Version
1.8.1.0

### Environment?
1. Hekate
2. Atmosphere 1.7.1 (official)
3. No relevant homebrew

### Additional context?
1. Region override worked on previous version of the game (1.0). If I remove the game and reinstall only the base game, behavior is correct
2. Setting language to Japanese in Ryujinx makes the game run in Japanese using 1.01 version
3. Tried adding the config.ini to 010026801E0AC800 and same behavior.

SciresM, 2024-07-15:
Post a screenshot of the config on the SD. This is almost certainly a user config issue.

fvig2001, 2024-07-16:
[image]
[image]

fvig2001, 2024-07-16:
Using config file above on 1.0
[image]

Using config file above on 1.01
[image]

SciresM, 2024-07-16:
That seems more than anything like the game is using a different method to determine locale than set calls? I genuinely think this isn't an atmosphere issue, if the game is using calls that we aren't intercepting. Feel free to talk more on discord?

closed 2024-07-16 by SciresM