Atmosphere-NX/Atmosphere #2363: PANIC OCCURRED WHILE RUNNING ATMOSPHERE ID: 010041544D530000

JohnMoreno04, 2024-07-25:
## Bug Report

### What's the issue you encountered?

Everything was going well, until when I passed a point necessary to progress the story in Zelda Tears of the Kingdom the atmosphere crashed, and this screen appeared. Whenever I restart it works normally, but whenever I enter the Zelda save and go through the same part it happens again.

[image]

I'm using CNX 18.1.0-3 and AMS 1.7.1

### How can the issue be reproduced?

</br>
Passing a certain point on the Zelda tears of the kingdom map. </br>

### Crash Report

[report_094cc3e7b.zip](https://github.com/user-attachments/files/16382418/report_094cc3e7b.zip)

### System Firmware Version

18.1.0</br>

### Environment?

Using hekate with TegraRCM
CNX 18.1.0-3
Atmosphere 1.7.1
- No kips or sysmodules
- Homebrew software installed: Some, but it came with the cnx
- EmuMMC SD File
- Fat 32
  

### Additional context?

- I have already downloaded two different zeldas believing that the previous file was corrupt, but that was not the case.
- I also used fix archive bit, but it didn't work either.
- reformatted the SD
- removed the ; from system_settings.ini, but not worked

masagrator, 2024-07-25:
```
I have already downloaded two different zeldas
```
Piracy is not supported.

borntohonk, 2024-07-26:
> * No kips or sysmodules
The "CNX" package you reference has 15 sysmodules...

JohnMoreno04, 2024-07-26:
> > * No kips or sysmodules
> 
> The "CNX" package you reference has 15 sysmodules...

Do you believe this could be the problem?
I'm just entering this world now, so I didn't know about sysmodules -_-

souldbminersmwc, 2024-08-02:
> ```
> I have already downloaded two different zeldas
> ```
> 
> Piracy is not supported.

I think they mean save files

closed 2025-04-03 by SciresM