Atmosphere-NX/Atmosphere #2366: atmosphere fatal error, program id: 0100000000005 (maybe enough zeros)

ooflol, 2024-08-01:
## Bug Report

### What's the issue you encountered?
i boot emummc and it has a fatal error

### How can the issue be reproduced?
from booting emummc from hekate

### Crash Report
there is none

### System Firmware Version
It is 18 as i just updated it
### Environment?
It was booted by hekate official v6.2.1, i use emummc

### Additional context?
I have android 10,

SciresM, 2024-08-01:
1) you didn't include the fatal error report
2) this is almost certainly emummc corruption, you'll probably need to reclone a fresh emummc,
3) this issue tracker is for bugs, not for support -- seek help with issues like this on discord

closed 2024-08-01 by SciresM

ooflol, 2024-08-01:
how do i reclone a fresh emummc, here is the fatal error https://www.mediafire.com/file/k63cc2rxjnrntbz/report_0000000000793b74.bin/file, which discord server?