Atmosphere-NX/Atmosphere #2369: Right Joy Con Lagging then disconnecting

Choccy-vr, 2024-08-11:
## Bug Report

### What's the issue you encountered?

While playing any game I found it to be most prominent in Mario kart it lags and then disconnects 

### How can the issue be reproduced?

Play any game wait a few minutes and the controller will lag and then disconnect

### Crash Report

N/A

### System Firmware Version

18.1.0|AMS 1.7.1|E
### Environment?

- What bootloader (fusèe, hekate, etc) latest version of hekate 
- Official release or unofficial build: 
official release 

- EmuMMC or SysNAND:
  EmuMMC partition based

masagrator, 2024-08-11:
Why did you delete point about used sysmodules? It's very important.

Choccy-vr, 2024-08-11:
> Why did you delete point about used sysmodules? It's very important.

I did not know I needed it. At one point I had sys-clk but I deleted it.

sthetix, 2024-11-09:
did it happen when docked? if docked then try to disable the usb3 function inside the atmosphere config

Choccy-vr, 2024-11-16:
Seems to be fixed on 1.8.0

closed 2024-11-16 by Choccy-vr