Atmosphere-NX/Atmosphere #2370: Atmosphere crashes everytime switch gives an error

GugsFr, 2024-08-14:
## Bug Report

### What's the issue you encountered?
Everytime the switch gives an error atmosphere crashes

### How can the issue be reproduced?
for example, when i try to connect to the internet and it fails
or when i try to go online but i havent payed for nintendo switch online
any kind of error

### Crash Report
https://gist.github.com/GugsFr/c6a4bf22ee9f4d096939748a8445d6aa

### System Firmware Version
16.1.0

### Environment?
- Hekate
- Official release 1.7.1
- SysNand

### Additional context?
according to https://switchbrew.org/wiki/Title_list the app id causing the error is reffered to "error (LibraryAppletError)" so its another thing that tells me that it crashes whenever an error is displayed

thank you and sorry if i forgot something

borntohonk, 2024-08-15:
> ### System Firmware Version
> 
> 16.1.0
> when i try to go online but i havent payed for nintendo switch online

Huh...?

Well nintendo much would prefer if you're on on the latest firmware, 18.1.0 if you want to actually "go online"

GugsFr, 2024-08-15:
As i said it was an example, i lost my original prodinfo so i cant go online even if i wanted to

larrrox, 2024-10-03:
this is probably because of your lack of prod info. this is just the error applet crashing perhaps because of too many error reports (no prodinfo). try setup dns.mitm here [https://switch.hacks.guide/extras/blocking_nintendo/](url), instead of naming the file `emummc.txt` name it `default.txt` as it'll fix the issue on both 'emummc' (if you have it) and 'sysnand'. let me know how it goes.

closed 2025-04-03 by SciresM