Atmosphere-NX/Atmosphere #2371: Atmosphere Always Crashing on Internet Menu

LuGB18, 2024-08-23:
## Bug Report
ERROR CODE: 2162-0002 (0x4a2)

### What's the issue you encountered?

Hi, Im trying to go on to the internet menu but it just keeps crashing,
i have some Plugins on but dont think that they are the problem, but i have some exefs-patches on for Splatoon 3
but i cant know if the problem is them.

### How can the issue be reproduced?

Just Acessing the internet menu while an game is opened (Splatoon 3 while acessed the SHOAL "Lan Play")

### Crash Report
[01724109140_0100c2500fc20000.log](https://github.com/user-attachments/files/16730475/01724109140_0100c2500fc20000.log)

### System Firmware Version

18.1.0

### Environment?

Using Hekate
Official Release (Atmosphere) 1.7.1
 Using SysPatches being sys-patch  
using EmuMMC (Partition-Based)

### Additional context?

N/A

Aura67, 2024-08-25:
Piracy is not supported and you use sys-patch that is already a sign of piracy,because of Piracy your atmosphere crashes.

LuGB18, 2024-08-25:
oh ok ill remove it

closed 2024-08-25 by LuGB18