Atmosphere-NX/Atmosphere #2372: Fusee Fatal Error: Unknown EmcDvfsTimingTableIndex: -1, Micron Dram

brxie, 2024-08-25:
## Bug Report

### What's the issue you encountered?

Launching atmosphere with fusee and with fss produces the error "Fusee fatal error: Unknown EmcDvfsTimingTableIndex: -1"
Console version Mariko with SN XTJ204...., memory vendor Micron, atmosphere 1.7.1 pure Hekate 6.2.0 pure

### How can the issue be reproduced?

N/A

### Crash Report

N/A

### System Firmware Version

17.0.0

### Environment?

- What bootloader (fusèe, hekate, etc.) was Atmosphère launched with: fusee and hekate
- Official release or unofficial build: Official release version 1.7.1
- Do you have additional system kips or modules that you are loading? No
- Homemade software installed: No
- EmuMMC or SysNAND: EmuMMC based on file.

### Additional context?

Dram ID MT53E512M32D1NP-046

[image]

SciresM, 2024-08-25:
Duplicate of #2354. This was fixed in the most recent atmosphere commit: https://github.com/Atmosphere-NX/Atmosphere/commit/4617fec672eb47744f05c20fdb26254cebd1f893#diff-b612219b1e78d1910d0db6c9d67ce8cbdc64344aac4a2111b64357964dedab7eR101

I guess a new build is needed, though, for you to use the fix.

closed 2024-08-25 by SciresM

brxie, 2024-08-26:
I saw this PR but I didn't realised it's also added for Micron DRAM. Thank you! :raised_hands: 
And yea, the new release would be very appreciated.