Atmosphere-NX/Atmosphere #2375: Creport fails to detect module symbol table for 17.0.0+ software

d2uxf0xs0jxgf3e, 2024-09-24:
As of 17.0.0, `.text+0x0` can be set to a value of `1` or `0x14000003`, which indicates that `.text+0x8` stores an offset to a 0xC-byte structure containing the SDK version major/minor/micro. The function `ModuleList::DetectModuleSymbolTable` in the creport implementation currently only checks if `.text+0x0` is set to a value of `0` or `0x14000002`, and returns early otherwise.

SciresM, 2024-09-24:
Does it really? Fascinating! What's a good example of software to verify this on?

d2uxf0xs0jxgf3e, 2024-09-24:
> What's a good example of software to verify this on?

I've confirmed it in at least World of Goo 2, which has an SDK version of 17.5.0 for all NSOs, but it's likely also present for any system processes/applets that were recompiled for 17.0.0.

SciresM, 2024-09-24:
Okay, verified. Unfortunately, fixing this is not quite as easy as it looks -- they now do a direct `adr x1, .dynamic`, which is problematic for us as it requires parsing aarch64 code. Going to look into whether there are other options.

SciresM, 2024-09-24:
Ah...you can just use MOD0 still. I am not sure why I wasn't doing that! I guess it saved an svc::ReadDebugProcessMemory read, but alas.

SciresM, 2024-09-24:
Fixed in 10c7a395285c133e4c17c52d84728a4c7b24accd, I think.