Atmosphere-NX/Atmosphere #2378: HBMenu applet mode softlocking Switch when playing Luigi's Mansion 3

Arales03482, 2024-09-29:
## Bug Report

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Console was getting softlocked every time (this issue did not happen on 13.x.x version where i downloaded the game through eshop with a older atmosphere ver)
All i had in atmosphere was tickets so i could launch games (yes i did buy them) and 
[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

1. Launch Luigi's Mansion 3 (1.4.0 preferably)
2. Open the hbmenu through applet mode
Now you have a softlocked console

### Crash Report

N/A

### System Firmware Version

16.0.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate v6.2.1
- Official release or unofficial build: 
  - v1.7.1 of Official Atmosphere
  - From Atmosphere-NX/Atmosphere and from the releases tab
- Do you have additional kips or sysmodules you're loading: not sure about kips and Noexes, sys-ftpd-10k, sys-clk, ldn_mitm (enabled), Tesla, emuiibo, MissionControl (i forgot what this does)
- Homebrew software installed: N/A
- EmuMMC or SysNAND:
  - EmuMMC file based

### Additional context?

- Additional info about your environment:
- Hosts file for emummc (its a 90dns hosts file)
- I have serial number blanking on emummc enabled

masagrator, 2024-09-29:
You have too much custom sysmodules. And that's probably why you have this issue.

Arales03482, 2024-09-29:
It works now
I think it was the MissionControl from the uLaunch pack?

closed 2024-09-29 by Arales03482

masagrator, 2024-09-29:
It doesn't matter. Issue was you had too much sysmodules. Deleting one was enough to free enough space for hbmenu to load.