Atmosphere-NX/Atmosphere #2379: 0x4A8(2168-0002)

Tanaka25809, 2024-09-30:
Bug Report
When playing Dragon Quest X Awakening Five Races Offline, it crashes at random times. If anyone knows what could be causing this, please let me know.
It seems to be related to the process name holiday, but I don't know what to do.

Crash Report
[01708690460_01006a800016e000.1.log](https://github.com/user-attachments/files/17188223/01708690460_01006a800016e000.1.log)

System Firmware Version
18.1.0

Environment?
Hekate 6.2.1
Official release 1.7.1
EmuNand

abhisheksonkar21, 2024-10-12:
[image]

Update this in json launch

Tanaka25809, 2024-10-12:
I'm sorry. Please let me know how to do it in detail and the operating procedure.

abhisheksonkar21, 2024-10-19:
Set the json value in your IDE between 8.9-6.96

closed 2025-04-03 by SciresM