Atmosphere-NX/Atmosphere #2384: Unable to Identify Package1 Issue

LachyTheMan, 2024-10-08:
## Bug Report

### What's the issue you encountered?

Whenever I try and boot into the fusee.bin file through hekate to launch atmosphere, the switch gives me an error saying "Unable to identify Package1."</br>

### How can the issue be reproduced?

</br>
So I was going to play Super Mario Bros. Wonder on my Switch, in Atmosphere. It then asked me for a system update, and at the time I did not realise that this would be like other consoles such as the Wii U, or 3DS and break the console if you do a system update. I have reinstalled all of the atmosphere, fusee, and sigpatch files multiple times, and even tried different versions of Atmosphere and the fusee payload file to see if it makes a difference. I have even gone to the extent of booting onto the vanilla switch os and factory resetting it. Still, when I go in to hekate and select the fusee payload, it still gives me the same issue. </br>

### Crash Report
N/A
There is no Crash Reports folder created, and no crash report file.

### System Firmware Version

19.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release 1.7.1, 1.6.2 or unofficial build: 
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ Yes ]
- EmuMMC or SysNAND: SysNAND

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

Masamune3210, 2024-10-08:
Piracy is not supported and you will get no help from here.

THZoria, 2024-10-08:
Effectively no piracy + Atmosphere does not support firmware 19.0.0

LachyTheMan, 2024-10-08:
> Effectively no piracy + Atmosphere does not support firmware 19.0.0

Ah ok. I'll see if I can find a way to downgrade my ofw back to 18.1.0. Thanks.

solleo1989, 2024-10-08:
> I also accidently updated Switch to 19.0.0. Will either try to live without OC for some time (until they update Atmosphere) or downgrade my switch back to 18.1.0. How long does it usually takes to update?

https://x.com/SciresM/status/1843443316446376123

LachyTheMan, 2024-10-08:
I managed to downgrade my ofw to 18.0.0. Problem solved.

closed 2024-10-08 by LachyTheMan

Masamune3210, 2024-10-10:
> Effectively no piracy + Atmosphere does not support firmware 19.0.0

Sigpatches = piracy. This isn't my words, this is a rule of the project as a whole.