Atmosphere-NX/Atmosphere #2386: 19.0.0 Support

SciresM, 2024-10-08:
Master issue/check list of stuff to do for 19.0.0 changes. Note that I can't actually work on this properly until Wed/Thurs, I expect

- [ ] Reverse engineer new kernel changes 
- [x] Update fusee/exosphere with new keydata/values
- [ ] Implement new changes into mesosphere
- [ ] Implement new changes into loader
- [ ] Implement new changes into pm
- [ ] Implement new changes into <sysmodules, create as needed>
- [ ] Update emummc
- [ ] Miscellaneous exosphere/fusee fixes for new DRAM IDs
- [ ] Make sure libnx is updated for anything relevant.

Post-release:
- [ ] Determine what other system modules need reverse engineering changes
- [ ] Reverse engineer potential new boot sysmodule changes.
- [ ] Miscellaneous post-release reverse engineering
- [ ] Miscellaneous post-release implementation work/updating

CoolCoderGuy, 2024-10-09:
Woah I did not realize this much went into getting cfw working on a new major firmware.

Visualwit, 2024-10-09:
Can't wait to get sleep mode working again on my oled's emummc!

steel101, 2024-10-09:
Sleep mode works on sysmmc on OLED at least on mine > Can't wait to get sleep mode working again on my oled's emummc!

xemdev, 2024-10-09:
Thanks for your work !!

ilykami, 2024-10-09:
how long do you think it would take to get the next build out to us? thanks alot for the work you do btw

masagrator, 2024-10-09:
> how long do you think it would take to get the next build out to us? thanks alot for the work you do btw

June 15... Jokes aside, never ask about ETA.

larrrox, 2024-10-09:
> Sleep mode works on sysmmc on OLED at least on mine > Can't wait to get sleep mode working again on my oled's emummc!
> 
> 

Does emummc work for you? I get the sleep warning on every boot since the update burned a fuse

ilykami, 2024-10-09:
> > how long do you think it would take to get the next build out to us? thanks alot for the work you do btw
> 
> June 15... Jokes aside, never ask about ETA. Follow official X account for such info https://x.com/sciresm

lmao okay thanks

SciresM, 2024-10-09:
This issue is for me to track my work, please don't spam it with comments.

jealion5, 2024-10-09:
Thanks for working on it! You're doing a great job!

prodlucidity, 2024-10-09:
Thank you for all the work you do :)

Naxdy, 2024-10-09:
Sorry for the post, but just FYI as org owner / admin you can lock the conversation to collaborators, this way you won't get annoy-pinged by random comments anymore. This is how it looks like on my repos:

[image]

Hopefully this will be your last ping :)

SciresM, 2024-10-09:
I am aware of how to lock issues, when I say "please do not spam", this includes e.g. advice comments on how to lock issues, thanks.

SciresM, 2024-10-10:
Kernel is almost-but-not quite fully updated now; I will be coming back to it to tackle the remaining work later. My TODO.txt contains:
```
* KPageTable implementation was heavily rewritten.
    * Too many changes to summarize here; the fundamental implementation is now based on iteration over levels using TraversalContext instead of separate per-level logic.
* Many heavy K(Initial)PageTable changes, including:
    * KInitialPageTable's table entries no longer have bit 58 (0x0400000000000000) set; previously this was used to indicate/determine whether a mapping was present even if the entry was NotMapped.
        * Bit0 (0x1) is now used to check for mapping existence again in many places.
    * KPageTableImpl::InitializeForKernel now iterates over mappings created by KInitialPageTable, setting PageAttribute=#used entries for tables and setting bit 58 on blocks.
```

Logic is that rewriting the page table implementation is likely to be much more error-prone for me-as-programmer, and it's a better prioritization of my time to get the rest of ams basically functioning before doing that.

SciresM, 2024-10-11:
Well, that's the kernel complete, unless there are bugs lurking in all that code I just wrote.

So far as I can tell, atmosphere is just working, but it's kind of hard for me to test something like this thoroughly. I uploaded a built to ReSwitched's discord, for people who are actually capable of helping me test that.

**Please only use this if you are competent enough to give me good bug reports**.

**Do not use this if you need  help with basic stuff/if you're not very familiar with atmosphere.**

**There may be hidden bugs lurking, please only use this if you're okay with your console being temporarily bricked if I fucked up something important.**

sand-undertale, 2024-10-11:
I'm new to the SwitchBrew scene, so 19.0.0 not being supported means that I won't be able to use SYS CFW?

Airfishgaming, 2024-10-11:
> I'm new to the SwitchBrew scene, so 19.0.0 not being supported means that I won't be able to use SYS CFW?

Pretty much, at least until it's released

elipltr, 2024-10-11:
> I'm new to the SwitchBrew scene, so 19.0.0 not being supported means that I won't be able to use SYS CFW?

you need to wait the official version of Atmosphere

zsdw666, 2024-10-11:
Silently pay attention to the boss and wait for the arrival of 19

SlicedHammm, 2024-10-12:
Any eta? No rush or anything just very impatient 😁

steel101, 2024-10-12:
Don't ask for ETA

SciresM, 2024-10-12:
I am going to give this a few extra days to simmer before doing a proper pre-release, however, things seem stable enough to me.

Please feel free to use this build, if you're interested in using atmosphere on 19.0.0 even before a pre-release: 

[atmosphere-1.8.0-preprerelease-5717ea6c0.zip](https://github.com/user-attachments/files/17348470/atmosphere-1.8.0-preprerelease-5717ea6c0.zip)
[fusee.bin.zip](https://github.com/user-attachments/files/17348471/fusee.bin.zip)

**Please note:** Usage of certain custom system modules (including sys-clk) can cause the system to hang when waking from sleep. This is not an atmosphere bug. Additionally, some sysmodules may need to be compiled with an upgraded switch-tools in order to function correctly, especially if they use debug SVCs.

SciresM, 2024-10-12:
I'm unlocking this issue, because I like looking at the emoji reactions to various posts. Any spam will be deleted, please do not make my life difficult as issue moderator, do not ask for help in this issue.

steel101, 2024-10-12:
Dumb question do you update firmware then atmosphere or the other way around

SlicedHammm, 2024-10-12:
How do I fix error 2144 on this build?

Peachywind, 2024-10-12:
love u

xdferpc, 2024-10-13:
In this version, it asks for 100% internet connection to start custom firmware (games) at launch

TouchedCoder, 2024-10-14:
Hope this isn't counting a spam sorry if it does Homebrew menu is crashing error code 2144-0001 (0x290) firmware 19.0.0 atmosphere 1.8.0 - 1900_support - 49763aee9-30
[image]

xdferpc, 2024-10-14:
> Hi. About atmosphere-1.8.0-preprerelease-5717ea6c0.zip
> fusee.bin.zip
>
> Unfortunately system crashes and no change compare to previous version.

It works fine for me, what happened was that I was missing the sign patches
for this version

Dmitry290378, 2024-10-14:
[image]

tsurugikengo, 2024-10-14:
> I am going to give this a few extra days to simmer before doing a proper pre-release, however, things seem stable enough to me.
>
> Please feel free to use this build, if you're interested in using [...]

Since it is not a bug in atmosphere, will the hang when waking from sleep still occur even after the official release?

evan-cloud, 2024-10-14:
I don't know if it's useful but it seems that these sigpatches work https://github.com/AK478BB/Sigpatches/releases 
Delete my comment if you don't find it relevant, apologies in advance

Skygames-yt, 2024-10-14:
i tried out the new pre-prerealease but when i try to go into the photo menu it keeps crashing with error 010000000000001000

Dmitry290378, 2024-10-14:
System crashes after few seconds after loaded.

Dmitry290378, 2024-10-14:
Hi. It crashes after few second when loaded. And no games starts, issuing
the errors. I don’t  understand why the posted this files, which killed all
the system.

SciresM, 2024-10-14:
I would like to be clear that any comment on this issue by another person is spam at this point, and I will begin banning people from the issue tracker if they continue not taking the hint.

SkippTekk, 2024-10-14:
Hi there, I have read above and willing to offer some assistance to what i can do... I'm very new to this but also willing to test/learn on how to help you improve. 

So, i have installed 19 on my friends switch (they are allowing me to crack it) and coming up with the following error [Image](https://imgur.com/M3MFFnc) hopefully this can help you know what's going on.

How i got it
Installed Hekate using switch hack guide website (version 6.2.2)
Used the above links for Atmos installation. [link](https://github.com/Atmosphere-NX/Atmosphere/issues/2386#issuecomment-2408306021)
Hekate Home > Launch > EMUMMC > Clicked on Album (which i understand is Homebrew?)

Version  from System Settings is 19.0.0|AMS 1.8.8|E

I have also noticed you said you would ban people.... I'm just wishing to assist and try out version if you need someone to do it.

Serial of switch XAQ1003226****

deadcuba, 2024-10-15:
Hola lo tengo funcional solo tengo error para ejecutar el TINFOIL por lo demas todo bien si les hace falta ayuda en algo o hacer pruebas tengo recursos

https://github.com/user-attachments/assets/944fc9c9-1be7-4ffc-b78f-61f8d543c71b

Obxndo, 2024-10-15:
> Hola lo tengo funcional solo tengo error para ejecutar el TINFOIL por lo demas todo bien si les hace falta ayuda en algo o hacer pruebas tengo recursos
> 
> 
> 
> 
> 
> https://github.com/user-attachments/assets/4dc2485d-04ee-4324-9eb6-84b7393bd094
> 
> 
> 
> 

Pues facilitar algún link para hacer una prueba?

WarioK, 2024-10-15:
> Hi there, I have read above and willing to offer some assistance to what i can do... I'm very new to this but also willing to test/learn on how to help you improve.
> 
> So, i have installed 19 on my friends switch (they are allowing me to crack it) and coming up with the following error [Image](https://imgur.com/M3MFFnc) hopefully this can help you know what's going on.
> 
> How i got it Installed Hekate using switch hack guide website (version 6.2.2) Used the above links for Atmos installation. [link](https://github.com/Atmosphere-NX/Atmosphere/issues/2386#issuecomment-2408306021) Hekate Home > Launch > EMUMMC > Clicked on Album (which i understand is Homebrew?)
> 
> Version from System Settings is 19.0.0|AMS 1.8.8|E
> 
> I have also noticed you said you would ban people.... I'm just wishing to assist and try out version if you need someone to do it.
> 
> Serial of switch XAQ1003226**** Edit: Sorry i found the [Logs](https://github.com/user-attachments/files/17371177/01728957901_0100000000001000.log)

Yes. for me homebrew channel is not working with this pre-pre-release version

bmichalkiewicz, 2024-10-15:
> Hi there, I have read above and willing to offer some assistance to what i can do... I'm very new to this but also willing to test/learn on how to help you improve.
>
> So, i have installed 19 on my [...]

"Do not use this if you need help with basic stuff/if you're not very familiar with atmosphere."

ReSwitched's Discord server is available for any user assistance.

impeeza, 2024-10-16:
> Hola lo tengo funcional solo tengo error para ejecutar el TINFOIL por lo demas todo bien si les hace falta ayuda en algo o hacer pruebas tengo recursos
> a.mp4

Piracy is NOT SUPPORTED!

DroidlyMobile, 2024-10-16:
Thank you daddy

ltecheroffical, 2024-10-16:
Also quick bug report, there is a hang when docking while in sleep mode.

bdawg1989, 2024-10-16:
> Also quick bug report, there is a hang when docking while in sleep mode.

This is not a bug. This is an issue with sys-clk that has been [addressed](https://github.com/retronx-team/sys-clk/commit/3ba7cbc11fe14e70a12d6b701916a2690b32af3b) already.

DroidlyMobile, 2024-10-16:
I found the issue with sleep mode as well, all I did was use the Atmo updater tool, updated everything including sig patches etc.  Now everything works fine even sysclk.

WarioK, 2024-10-16:
> "Do not use this if you need help with basic stuff/if you're not very familiar with atmosphere."
>
> ReSwitched's Discord server is available for any user assistance.

No idea what that users experience is.  But I am very familiar with Atmosphere, and the issue they reported is legitimate.

ltecheroffical, 2024-10-16:
> > Also quick bug report, there is a hang when docking while in sleep mode.
> 
> This is not a bug. This is an issue with sys-clk that has been [addressed](https://github.com/retronx-team/sys-clk/commit/3ba7cbc11fe14e70a12d6b701916a2690b32af3b) already.

Then explain how it still is doing it after sys-clk was disabled?

SciresM, 2024-10-16:
It is amazing how many people commented on this issue after I said "nobody else should comment on this issue".

ltecheroffical, 2024-10-16:
> It is amazing how many people commented on this issue after I said "nobody else should comment on this issue".

Here is one guess, you basically told the internet what to do and the moment you (or anyone) do that, there is at least one person who will start a chain of events doing the opposite of what you want. There is barely a way to stop this.

SciresM, 2024-10-16:
That guy has been blocked from interacting with this repo for 7 days. This will also happen to anyone else who does a stupid, here.

Neomaster397, 2024-10-17:
Hey, calm down friend... the majority just want to help and contribute. I don't know why they are so unpleasant... I think you missed the class where they teach you that it is a free country with the right to expression, you shouldn't act like a dictator. Most of them do not do it with bad intentions, we just want to help, because we need and want the complete update of this

devgeda, 2024-10-17:
Anyone who has problems and wants to contribute, just start a new issue and then help. now your comment will probably generate a herd effect and a lot of people will label him a dictator or something like that, don't generate a discussion about a problem that shouldn't exist, it was made very clear not to comment because this issue is ONLY to monitor progress , if you have a problem, start another ISSUE. I understand that my comment will, and should, be removed. Don't be unnecessary man, don't create disorder because of this, it's been made clear more than once and those who don't respect it are either because they can't understand a simple rule to maintain order or they don't care. I authorize my comment to be removed and I support, please, not leaving this ISSUE JUST TO MONITOR PROGRESS AND NOT AN ISSUE TO RESOLVE PROBLEMS! in order.

Dicekaurora, 2024-10-18:
not sure why it wont boot but here is the crash bin.  Crashes after the nintendo logo
[report_00000000096aa754.zip](https://github.com/user-attachments/files/17427893/report_00000000096aa754.zip)

SkippTekk, 2024-10-18:
> not sure why it wont boot but here is the crash bin. Crashes after the nintendo logo [report_00000000096aa754.zip](https://github.com/user-attachments/files/17427893/report_00000000096aa754.zip)

Go to [here](https://github.com/Atmosphere-NX/Atmosphere/issues) and create a new issue. This is for UPDATES only as stated above.

fasajr, 2024-10-20:
la consola se apaga en lugar de suspenderse, tambien he notado que en algunas consolas se descalibra la bateria, tengo 3 switch y dos presentan problemas

Dicekaurora, 2024-10-20:
i deleted the atmosphere.folder and reinstalled added it again.  it is
working fine now thanks.. even mario party jamboree.  sucks i had to format

bulzipke, 2024-10-25:
[image]

I am using the latest libctru.
When I build, I get a static_assert.

It seems to be caused by 16 bytes of unused_20 in the ProgramInfo, which seems to be a different size than the LoaderProgramInfo in [libnx's struct.](https://switchbrew.github.io/libnx/structLoaderProgramInfo.html) 

Sorry to bother you, but do I need to use a specific version or make a modification to get it to build?

dracula99gr, 2024-10-30:
Good morning @SciresM! keep in mind even newer fw got released, 19.0.1. Thank you for all you have done and thank you for your patience!

SciresM, 2024-10-30:
Atmosphere 1.8.0 supports 19.0.1 with no changes. There is no need to comment on this issue.

[43 more replies]