Atmosphere-NX/Atmosphere #2388: Host file not being applied

steel101, 2024-10-09:
## Bug Report

### What's the issue you encountered?

I installed atmosphere in two different switches with all the newest stuff and when I apply the host file to block Nintendo servers Nintendo's eShop is blocked but it's still able to download from nintendo.net for the system update files so it's not true  blocking it but if I add the same host file to my private next DNS and hotspot from my phone to the switch it can't get a data connection to Nintendo servers at all so I know the host file is not being applied correctly

### How can the issue be reproduced?

Install atmosphere like normal and apply the host file to block Nintendo servers

### Crash Report

N/a

### System Firmware Version

18.0.1

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: moded OLED 
- Official release or unofficial build: 
  - latest release
     This github
- Do you have additional kips or sysmodules you're loading:no
- Homebrew software installed: [ * ]
 SysNAND:
 

### Additional context?

- Additional info about your environment:
- n/a

masagrator, 2024-10-09:
Maybe upload this host file? :)

steel101, 2024-10-09:
# 90DNS-equivalent
127.0.0.1 *nintendo.com
127.0.0.1 *nintendo.net
127.0.0.1 *nintendo.jp
127.0.0.1 *nintendo.co.jp
127.0.0.1 *nintendo.co.uk
127.0.0.1 *nintendo-europe.com
127.0.0.1 *nintendowifi.net
127.0.0.1 *nintendo.es
127.0.0.1 *nintendo.co.kr
127.0.0.1 *nintendo.tw
127.0.0.1 *nintendo.com.hk
127.0.0.1 *nintendo.com.au
127.0.0.1 *nintendo.co.nz
127.0.0.1 *nintendo.at
127.0.0.1 *nintendo.be
127.0.0.1 *nintendods.cz
127.0.0.1 *nintendo.dk
127.0.0.1 *nintendo.de
127.0.0.1 *nintendo.fi
127.0.0.1 *nintendo.fr
127.0.0.1 *nintendo.gr
127.0.0.1 *nintendo.hu
127.0.0.1 *nintendo.it
127.0.0.1 *nintendo.nl
127.0.0.1 *nintendo.no
127.0.0.1 *nintendo.pt
127.0.0.1 *nintendo.ru
127.0.0.1 *nintendo.co.za
127.0.0.1 *nintendo.se
127.0.0.1 *nintendo.ch
127.0.0.1 *nintendo.pl
127.0.0.1 *nintendoswitch.com
127.0.0.1 *nintendoswitch.com.cn
127.0.0.1 *nintendoswitch.cn
95.216.149.205 *conntest.nintendowifi.net
95.216.149.205 *ctest.cdn.nintendo.net

steel101, 2024-10-09:
Direct copy of the host file

masagrator, 2024-10-09:
I explicitly said "file", not its contents.

masagrator, 2024-10-09:
The file you have on your sdcard right now

steel101, 2024-10-09:
[image]

steel101, 2024-10-09:
[emummc.txt](https://github.com/user-attachments/files/17304968/emummc.txt)

larrrox, 2024-10-09:
By any chance are you booted into syscfw and not emummc?.. If for some reason you want to block Nintendo on syscfw too you have to rename the file to default.txt

steel101, 2024-10-09:
[default.txt](https://github.com/user-attachments/files/17309703/default.txt)
[sysmmc.txt](https://github.com/user-attachments/files/17309704/sysmmc.txt)
[emummc.txt](https://github.com/user-attachments/files/17309706/emummc.txt)

steel101, 2024-10-09:
If you try to use the web browser Nintendo is blocked and using 9dnstester shows it is blocked but using tcp monitoring shows it is downing data from nintendo.net for firmware update would like that to be blocked so added the direct link for the update server and it is still not blocked in the host file

steel101, 2024-10-09:
[image]

steel101, 2024-10-09:
[image]

steel101, 2024-10-09:
Using netcat to watch network traffic and Nintendo's serves still a connected vs going to local host

steel101, 2024-10-09:
V1 switch is emummc and OLED switch is sysmmc  so I have all the same files on both switch and they both are able to connect to Nintendo servers so it not matter if it is sysmmc or emummc

SciresM, 2024-10-09:
set atmosphere!enable_dns_mitm_debug_log = u8!0x1 in settings, then run ams and post the atmosphere/logs/dns_mitm_debug.log that's created.

steel101, 2024-10-09:
How do I set that where is it located at in a file im guessing

SciresM, 2024-10-09:
Copy system_settings.ini from /atmosphere/config_templates to /atmosphere/config, uncomment the relevant line by removing the semicolon at start of line here: https://github.com/Atmosphere-NX/Atmosphere/blob/master/config_templates/system_settings.ini#L62 and change the value at the end of the line to u8!0x1

SciresM, 2024-10-09:
ams is atmosphere, the software you're commenting on the issue tracker for.

steel101, 2024-10-09:
Ok give me a 30 mins or so and I will.be home to charge settings and try again

steel101, 2024-10-09:
[dns_mitm_startup.log](https://github.com/user-attachments/files/17311800/dns_mitm_startup.log)

steel101, 2024-10-09:
The debug one I can't copy because it is 0b

steel101, 2024-10-09:
[image]

steel101, 2024-10-09:
This was after I enabled debug that you asked for it is still connected to Nintendo

steel101, 2024-10-09:
I fully understand 19 is more important. Like I said it blocks all eShop and game updates but allows firmware updates to go thru

steel101, 2024-10-09:
I blank my serial number out then Nintendo servers can't connect

steel101, 2024-10-09:
The reason I guess most people don't notice is unless there was a firmware update then you would be blocking Nintendo for them but as you see in the screenshot there is lots of connections to Nintendo servers

Coxxs, 2024-10-11:
If you set "Proxy Server" in the Switch OS, `DNS.mitm` (hosts file) becomes useless because DNS resolution is done through your proxy server.

cucholix, 2024-10-12:
I’m on 18.0.1 too, default.txt file is effectively blocking Nintendo updates, note that I **only** blocked system updates.

My default.txt:
`sun.hac.%.d4c.nintendo.net
atumn.hac.%.d4c.nintendo.net
127.0.0.1 sun.hac.lp1.d4c.nintendo.net 
127.0.0.1 atumn.hac.lp1.d4c.nintendo.net`

My dns_mitm_startup.log report:
`DNS Mitm:
Adding defaults to redirection list.
Selecting hosts file...
Skipping /hosts/sysmmc.txt because it does not exist...
Selected /hosts/default.txt
Redirections:
    atumn.hac.lp1.d4c.nintendo.net -> 127.0.0.1
    sun.hac.lp1.d4c.nintendo.net -> 127.0.0.1
    receive-lp1.er.srv.nintendo.net -> 127.0.0.1
    receive-lp1.dg.srv.nintendo.net -> 127.0.0.1`
    
It successfully blocked Nintendo system updates, I can enter eShop, play online, and even dowload games, don't have game updates pending but I guess they should work too.

These settings in `system_settings.ini`
`enable_dns_mitm = u8!0x1
enable_dns_mitm_debug_log = u8!0x1`

harvestry-of-ghosts, 2024-10-21:
> # 90DNS-equivalent
> 127.0.0.1 *nintendo.com 127.0.0.1 *nintendo.net 127.0.0.1 *nintendo.jp 127.0.0.1 *nintendo.co.jp 127.0.0.1 *nintendo.co.uk 127.0.0.1 *nintendo-europe.com 127.0.0.1 [...]

You can shrink your list a good amount by just replacing a lot of those with a single instance of:
127.0.0.1 *nintendo.*

harvestry-of-ghosts, 2024-10-21:
127.0.0.1 *nintendo.com
127.0.0.1 *nintendo.net
127.0.0.1 *nintendo.jp
127.0.0.1 *nintendo.co.jp
127.0.0.1 *nintendo.co.uk
127.0.0.1 *nintendo-europe.com
127.0.0.1 *nintendowifi.net
127.0.0.1 *nintendo.es
127.0.0.1 *nintendo.co.kr
127.0.0.1 *nintendo.tw
127.0.0.1 *nintendo.com.hk
127.0.0.1 *nintendo.com.au
127.0.0.1 *nintendo.co.nz
127.0.0.1 *nintendo.at
127.0.0.1 *nintendo.be
127.0.0.1 *nintendods.cz
127.0.0.1 *nintendo.dk
127.0.0.1 *nintendo.de
127.0.0.1 *nintendo.fi
127.0.0.1 *nintendo.fr
127.0.0.1 *nintendo.gr
127.0.0.1 *nintendo.hu
127.0.0.1 *nintendo.it
127.0.0.1 *nintendo.nl
127.0.0.1 *nintendo.no
127.0.0.1 *nintendo.pt
127.0.0.1 *nintendo.ru
127.0.0.1 *nintendo.co.za
127.0.0.1 *nintendo.se
127.0.0.1 *nintendo.ch
127.0.0.1 *nintendo.pl
127.0.0.1 *nintendoswitch.com
127.0.0.1 *nintendoswitch.com.cn
127.0.0.1 *nintendoswitch.cn
95.216.149.205 *conntest.nintendowifi.net
95.216.149.205 *ctest.cdn.nintendo.net

This list is so long. Can replace most of it with a wild card like this?

127.0.0.1 *nintendo.*
127.0.0.1 *nintendods.*
127.0.0.1 *nintendoswitch.*
95.216.149.205 *nintendowifi.*
95.216.149.205 *ctest.cdn.nintendo.net

Except the last one:
95.216.149.205 *ctest.cdn.nintendo.net

Would that not already be covered by:
127.0.0.1 *nintendo.* ???

If so, does it then redirect to 127.0.0.1 or 95.216.149.205?

steel101, 2024-10-21:
The problem is it effectively looks like it's blocking all the connections on the device but if you use a netcat monitor or a TCP IP packet inspector on a different device for over your Wi-Fi you'll still see a ton of the connections that are connecting and successful to the Nintendo servers using the DNS private part not the DNS we can only block DNS right now AMS needs to update to include the dns priv

steel101, 2024-10-21:
I have to block them on my DNs server for the device to not connect

SciresM, 2024-10-21:
That isn't actually the problem, fyi. dns:priv is completely unused; nothing has access to it @steel101 

I strongly suspect at this point that this is some issue with how you're testing, not with the software.

SciresM, 2024-10-21:
@steel101 When you use something like PCAPdroid to test, are you setting it as a proxy on your console?

because this will make your phone do dns resolution, not the console, which would bypass dns mitm.

steel101, 2024-10-21:
I use a TCP IP monitor so I can see all connections that are active and there is lots that gets by the DNS rules in the host file.

SciresM, 2024-10-21:
"use a TCP IP monitor" what monitor and how? You posted screenshots of pcapdroid and didn't respond to my question just now.

steel101, 2024-10-21:
I have pi hole set up too and all the same connections are there and no I am not doing a proxy on the console I'm just sniffing the Wi-Fi packets at that point with that app no proxy needed

steel101, 2024-10-21:
> "use a TCP IP monitor" what monitor and how? You posted screenshots of pcapdroid and didn't respond to my question just now.

I use pie hole I've used netcat and the app unused all show the same thing

SciresM, 2024-10-21:
This is kind of a big shrug for me. I can't replicate this locally, and I don't understand networking well enough to do anything more. It certainly seems to me like blocking works on my device...

steel101, 2024-10-21:
You can use netcat or just your router dchp list and look through there and see all the active Nintendo connections once you connect a Nintendo switch to your Wi-Fi so I'm just being honest and truthful that I work in the IT industry and I can tell you the host file is not blocking everything that it should

steel101, 2024-10-21:
> This is kind of a big shrug for me. I can't replicate this locally, and I don't understand networking well enough to do anything more. It certainly seems to me like blocking works on my device...

Yes you would be correct it seems like it's being blocked on the device and all the actual main servers like the eShop and updates are but there's a ton of telementary data about what game is being played how long and all the other things that happen on the console are sent back to Nintendo every 3 minutes so that's not being blocked but it's no big deal really I mean if they have sort of banning thousands of consoles by now I don't think they will

steel101, 2024-10-21:
And the way that you can replicate it is use your Android or iPhone to create a hotspot install the app that you saw or netcat and monitor it and connect the switch to your Wi-Fi hotspot through your phone and watch all the active connections very easy to replicate

harvestry-of-ghosts, 2024-10-21:
> > This list is so long. Can replace most of it with a wild card like this?
> 
> You can, but then those wildcards might also match part of a subdomain. E.g. _somethingabout_**_nintendo._**_in-some-site-you-actually-want-to-visit.whatever_
> 
> > If so, does it then redirect to 127.0.0.1 or 95.216.149.205?
> 
> [If multiple entries in a host file match a domain, the last-defined match is used.](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md#hosts-files)

I'll keep my short list then, as I don't want or need my emuMMC to connect to anything "Nintendo" whatsoever and it is set up only on my Switch. Thanks for the explanation/link on the domains. That brings up another question:

127.0.0.1 *nintendo.*
vs
95.216.149.205 *nintendo.*

What would be the advantages/disadvantages of one over the other? Why not use localhost for all entries in the hosts file?

steel101, 2024-10-21:
Nintendo has not don't nothing about at this point they will not I have checked this on over 8 times on different switches and everything is the same on the console it is blocked but behind the apps the user sees it is connected to Nintendo but like I said they have not done anything at this point for this long they can't tell it apart from.a real switch is my guess

steel101, 2024-10-21:
Been messing with all day trying different things what I found to block it with no connection to Nintendo's servers is take out 
95.216.149.205 *conntest.nintendowifi.net
95.216.149.205 *ctest.cdn.nintendo.net
And if it is not host file is not named default.txt or you deleted you default.txt file it will not block all connections. So what I did is replace default.txt with the updated one

impeeza, 2024-10-21:
@SciresM  There is a rummor on forums about nintendo using telemetry to IPs instead of names rendering DNS blocking mute. Have you seen something like that on the new firmware code?

Regards.

steel101, 2024-10-22:
> @steel101 
> 
> My default.txt is unchanged. My emummc.txt is as follows and should cover all entries in the long list. Maybe can even remove the "." at the end of each line before the wildcard.
> 
> ```
> 127.0.0.1 *nintendo.*
> 127.0.0.1 *nintendo-europe.*
> 127.0.0.1 *nintendowifi.*
> 127.0.0.1 *nintendods.*
> 127.0.0.1 *nintendoswitch.*
> ```
> I believe the 95.216.149.205 entries are not necessary if a person has ctest patched to skip the connection tests. It is also recommended to keep the default.txt as is and also have emummc.txt and/or sysmmc.txt. https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/features/dns_mitm.md#atmosph%C3%A8re-defaults
> 
> Edit: Wouldn't this also work for a person using the hosts file strictly on their Switch? Just a single line in emummc.txt?
> 
> `127.0.0.1 *nintendo*`
> 
> Wouldn't that include everything on the long list with a single line?
> 

You are correct about  the last 2 lines in the file no need for them and about the 1 line method I think that might work I will try it latter

steel101, 2024-10-22:
> @SciresM  There is a rummor on forums about nintendo using telemetry to IPs instead of names rendering DNS blocking mute. Have you seen something like that on the new firmware code?
> 
> Regards.

As long as the DNS is blocking Nintendo you are fine it is a rumor for a reason can you telementary data with out sending it to Nintendo so we can always block that data flow

closed 2025-04-03 by SciresM