Atmosphere-NX/Atmosphere #2391: Booting to black screen in pre-prerelease

gainb, 2024-10-12:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
Updated my switch FW. Then updated Atmosphere and Hekate
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>
No

### How can the issue be reproduced?
N/a

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>
N/a
### Crash Report
N/a
[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version
19.0.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - Pre pre release for 19.0.0
- Do you have additional kips or sysmodules you're loading:
- No
- Homebrew software installed: [ N/a ]
- EmuMMC or SysNAND:
  - SysNAND

pegast73, 2024-10-12:
**There is no pre-release yet for fw 19.0.0** !

SciresM, 2024-10-12:
This is definitely a user issue, black screen basically always are these days. Anyway, pre-prerelease is for support on discord, not github.

closed 2024-10-12 by SciresM