Atmosphere-NX/Atmosphere #2393: Atmosphere crashes upon loading emuMMC

Rodostlyn69, 2024-10-14:
[01728901942_010000000000000c.log](https://github.com/user-attachments/files/17363159/01728901942_010000000000000c.log)

Also, the contents folder in Atmosphere is missing and the folder called 010000000000000C doesn't exist on the SD

masagrator, 2024-10-14:
Fill the issue template that you removed.

BaamAlex, 2024-10-20:
I comment here because I don't want to open a new issue. I have technically the same problem. I noticed that today after updating atmosphere to 1.8.0 pre-release. Firmware is on 19.0.0 on both systems. I played Zelda echoes of wisdom, paused the game, pressed the home button and let the console idle for a few minutes and before I noticed anything, my console restarted into my sysnand. And this happened twice or more today.

n3vros, 2024-10-21:
same issue here

SciresM, 2024-10-24:
This is just a corrupt emummc, not an atmosphere bug

closed 2024-10-24 by SciresM

BaamAlex, 2024-10-24:
> This is just a corrupt emummc, not an atmosphere bug

What now? Mine or do you mean the OP?

SciresM, 2024-10-24:
the issue tracker is for bugs, not for support. Seek support on discord.