Atmosphere-NX/Atmosphere #2394: sleep mode error

Valou5231, 2024-10-16:
## Bug Report

### What's the issue you encountered?

after  i have update Atmosphere the bug has spawn

### How can the issue be reproduced?

whene i turn my switch into sleep mode and  i want to turn on it , the switch reboot on hekate.

### System Firmware Version

19.0.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

Atmosphere was lauched bye hekate on the v6.2.2

ELY3M, 2024-10-16:
Do you have sys-clk or any sys-modules?       make sure you update all of your sys-modules.

Valou5231, 2024-10-16:
yes i have a sys-modules and all is update

redilsnn, 2024-10-17:
Having this same problem after updating, but for me it totally blacks out when trying to wake up, and the only option Is holding the power button down for several seconds to shut down the consoles completetely, which Is annoying because i have to re-do the joycon jig and send payload with tegra everytime to boot into hekate 

Will the "Turning on AutoRCM" option serve as a workaround in the meantime? (For me to not do the jig thing everytime this happens?)

Valou5231, 2024-10-17:
me too the switch is totally off once in two

Yellowpandasomething, 2024-10-17:
I am also having the same issue as redilsnn where once I put it to sleep the system becomes completely unresponsive and I'm forced to force shut it down and rejig it. I'm on atmosphere 1.8.0, switch os 19.0.0, and hekate 6.2.2.

Fa1lnaught, 2024-10-17:
Same here and when I try to boot into OFW I get warning saying "unknown pkg1 version hos version not supported"

whossein, 2024-10-17:
I also have a problem on my device, OLED

masagrator, 2024-10-17:
sys-clk is not the only tool that you must update after updating to atmosphere 1.8.0 to not have sleep issues.

List of tools that can cause this issue:
- sys-clk (fixed the issue in newest release)
- Status Monitor Overlay (fixed the issue in newest release)
- Edizon Overlay (not fixed, don't go to Stats menu if you don't want this to happen)
- nx-fancontrol (was not fixed yet)
And any other tool that reads temperatures for whatever reason.

masagrator, 2024-10-17:
> when I try to boot into OFW I get warning saying "unknown pkg1 version hos version not supported"

You didn't update Hekate.

Update everything folks, this is the best advice whenever new firmware is released.

Fa1lnaught, 2024-10-17:
Yeah this is the first time I've gotten CFW on something so I don't know much

Valou5231, 2024-10-18:
ok i wil try this

closed 2024-10-20 by SciresM

ToxiClay, 2024-11-04:
Cheers to @masagrator for saving my dumb ass from opening a useless, redundant issue. Updating sys-clk fixed it.

javoerro, 2024-12-03:
im having the same issue on 19.0.1. OFW works fine, the problem is in CFW and EMU.

ToxiClay, 2024-12-04:
> im having the same issue on 19.0.1. OFW works fine, the problem is in CFW and EMU.

Update your shit.

javoerro, 2024-12-04:
ISSUE UPDATE: updating sys-clk homebrew solved my problem!