Atmosphere-NX/Atmosphere #2395: Error 2168-0002 on pre-release 18.00

sand-undertale, 2024-10-16:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Whenever I boot up CFW (sysMMC) it spits out error code 2168-0002.

### How can the issue be reproduced?

Booting up CFW.

### Crash Report

https://gist.github.com/sand-undertale/22625dcc4161a9d5e4881c2931d4d380

### System Firmware Version

19.0.0

### Environment?

Hekate with RetroArch and Tinfoil.

SysNAND

Aura67, 2024-10-16:
Piracy is not supported

sand-undertale, 2024-10-16:
huh

sand-undertale, 2024-10-16:
i dont pirate with tinfoil, i use cheat codes

TheButterGoblin, 2024-10-16:
I have 2 questions. 1. by environment do you want me to delete those or does it mean there is a problem from them causing the issues 2. do you want us to install that text document or not. (I'm guessing not but just making sure) please get back to me soon

sand-undertale, 2024-10-16:
by environments i mean they are causing issues, i believe that atmosphere has an issue, and it wants me to paste the crash log, so thats where the file came from.

TheButterGoblin, 2024-10-16:
so I deleted retro arch the Hekate bin file from within my files along with tinfoil the app and the NSP file and that didn't work so my problem hasn't been fixed. now my 2 final questions 1. how do I copy the error problem or would I copy yours. 2.  did it work when you tried these fixes.

wakoyname, 2024-10-19:
then downgrade is your only option. you can still play games on fw version 18.1.0 using atmos 1.7.1, why update?

closed 2024-10-20 by sand-undertale