Atmosphere-NX/Atmosphere #2397: Error Code 2123-0011 (0x167b)

jamfrancisc0, 2024-10-17:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Once updating atmosphere and booting up an existing emunand it crashes withing a few seconds with error code
2123-0011, ive tried twice now it happened the same both times

### How can the issue be reproduced?

[ * ]</br>
boot into hekate and click on emummc then just wait

### Crash Report

(https://gist.github.com/jamfrancisc0/8fc6b70846a23df3035b0f40ea4e2f85)

### System Firmware Version

19.0.0

### Environment?

used hekate to load a partition based emummc

### Additional context?

borntohonk, 2024-10-18:
@jamfrancisc0 see https://github.com/Atmosphere-NX/Atmosphere/issues/2225
The solution is for you to either reverse engineer NIM and create patch that removes the fatal condition, or...

edit exosphere.ini like this, to whichever line you have set to 1. (this is not new in 19.0.0, this has been like this since 17.0.0)
blank_prodinfo_sysmmc=0
blank_prodinfo_emummc=0

jamfrancisc0, 2024-10-18:
@borntohonk both are already at 0 in exosphere.ini, cant find cal0blank in hekate_ipl should i create it?

borntohonk, 2024-10-18:
> @borntohonk both are already at 0 in exosphere.ini, cant find cal0blank in hekate_ipl should i create it?

You probably used "incognito" at some point then, which is irreversible without a backup of your prodinfo

wakoyname, 2024-10-18:
also delete atmosphere > hosts > default.txt

jamfrancisc0, 2024-10-18:
@borntohonk @wakoyname nothing seems to work, theres nothing even in atmosphere>config and ive deleted default.txt but no joy, any ideas to fix the problem thats not above my pay grade as im no expert? if i wipe the sd and start a new emunand will it have the same fault?

borntohonk, 2024-10-18:
> @borntohonk @wakoyname nothing seems to work, theres nothing even in atmosphere>config and ive deleted default.txt but no joy, any ideas to fix the problem thats not above my pay grade as im no expert? if i wipe the sd and start a new emunand will it have the same fault?

I've already responded to you and provided you your realistic options.

No need to tag me further, do with the options you have.

Undo incognito/incognito_rcm, or, alternatively, reverse engineer a patch for NIM to remove the fatal (repeated for every firmware version released)

a new emunand would have the same occur if your "sysnand" is where you applied "incognito".

jamfrancisc0, 2024-10-18:
@borntohonk sorry i wont bother you anymore i just have no idea how to even begin reverse engineering a patch so i was hoping for another solution. thanks for the help anyway :)

wakoyname, 2024-10-19:
then downgrade is your only option. you can still play games on fw version 18.1.0 using atmos 1.7.1, why update?

closed 2024-10-20 by SciresM