Atmosphere-NX/Atmosphere #2398: Atmosphere crashes 1.8.0 crashes when connecting to wi fi

LeilaFlowers, 2024-10-17:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

Connect to wi fi in atmosphere 1.8.0, emummc ver 19.0.0

### Crash Report

https://github.com/user-attachments/assets/ae535f48-c514-43c7-bff8-2fa265eec94b

### System Firmware Version

Emummc 19.0.0

### Environment?

Hekate lastest

228hal, 2024-10-18:
Try disabling blanked product ID in exosphere if you have it enabled

Ogurez123, 2024-10-18:
This is most likely due to the fact that your nintendo switch is trying to connect to the nintendo servers. Follow these instructions https://switch.hacks.guide/extras/blocking_nintendo /
it helped me.

wakoyname, 2024-10-18:
if both are already at 0 in exosphere.ini:

try to check if theres another exosphere.ini in root folder and delete it
go to atmosphere > config > system_settings
find: enable_dns_mitm = u8!0x1
and change it to: ; enable_dns_mitm = u8!0x1
find: add_defaults_to_dns_hosts = u8!0x1
and change it to: ; add_defaults_to_dns_hosts = u8!0x1

go to atmosphere > config_templates > system_settings
find: enable_dns_mitm = u8!0x1
and change it to: ; enable_dns_mitm = u8!0x1
find: add_defaults_to_dns_hosts = u8!0x1
and change it to: ; add_defaults_to_dns_hosts = u8!0x1

also delete atmosphere > hosts > default.txt

LeilaFlowers, 2024-10-18:
Is incógnito mode not working then?

Masamune3210, 2024-10-19:
it hasnt for at least two updates. It's not that it doesn't work, its that the Big N started specifically checking for issues with getting the information and the system panics if the information is missing or modified

wakoyname, 2024-10-19:
then downgrade is your only option. you can still play games on fw version 18.1.0 using atmos 1.7.1, why update?

Masamune3210, 2024-10-20:
risky as in you might have to start fresh and wipe all your saves and profiles, not risky as in brick

closed 2024-10-20 by SciresM

C1p86, 2024-11-05:
hello! I have the same problem, a switch that was updated to 19.0.1 and now downgraded to 18.0.1.
It crashes as it connect to wifi, but mitm is working (in the log I see redirects enabled). Any clues?

ZeroSynn-zip, 2025-05-11:
Do i disable blank product id of sysmmc or emummc