Atmosphere-NX/Atmosphere #2399: All installed games give 'Unable to start software'

TroubledGamer, 2024-10-18:
I am on FW 19.0.0 and atmosphere 1.8.0
Everything was working fine until the update.. Not sure what to do.

masagrator, 2024-10-18:
1. You didn't use issue template.
2. Reinstall games from eshop.

TroubledGamer, 2024-10-19:
> 1. You didn't use issue template.
> 
>     2. Reinstall games from eshop.

I am afraid this does nothing.. Every single game I've tried is unable to open.

Ogurez123, 2024-10-19:
If your games are pirated, then most likely there is a problem with sigpatches.
Try using this archive with gbatemp 
https://gbatemp.net/threads/sigpatches-for-atmosphere-hekate-fss0-fusee-package3.571543/post-10517797

HunterMario, 2024-10-19:
Piracy is not supported.

borntohonk, 2024-10-19:
> > 1. You didn't use issue template.
> >    
> >    2. Reinstall games from eshop.
> 
> I am afraid this does nothing.. Every single game I've tried is unable to open.

He told you to go to eshop and pay for the games, the ones not opening are games you didn't pay for.

Piracy is not supported.

closed 2024-10-20 by SciresM