Atmosphere-NX/Atmosphere #2400: console turns off instead of suspending

Julio1322134388, 2024-10-19:
When you use the power button to suspend the console, the Switch turns off instead.

borntohonk, 2024-10-19:
> When you use the power button to suspend the console, the Switch turns off instead.

didn't fill out the template you deleted when making this issue, also no it doesn't. That's probably something on your end in your atmosphere/contents/ folder

MantleGaming, 2024-10-19:
This happens to me as well. I have nothing in my contents folder. It’s very annoying

paulzuri, 2024-10-19:
having the same issue here on firmware 19.0.0

borntohonk, 2024-10-19:
> This happens to me as well. I have nothing in my contents folder. It’s very annoying

in https://github.com/Atmosphere-NX/Atmosphere/issues/2386#issuecomment-2408306021

> Please note: Usage of certain custom system modules (including sys-clk) can cause the system to hang when waking from sleep. This is not an atmosphere bug. Additionally, some sysmodules may need to be compiled with an upgraded switch-tools in order to function correctly, especially if they use debug SVCs.

"sysmodules" and such, exist in the /atmosphere/contents/ folder, as do game mods, "custom themes", etc.

You you insist you have this problem with "nothing" in that folder, then that'd be weird that you have an issue caused by something that lives in the /atmosphere/contents/ folder

You aren't lying/providing bad information here right? Because that benefits no one to give inaccurate information

closed 2024-10-20 by SciresM