Atmosphere-NX/Atmosphere #2402: Not able to suspend on emuMMC, turning off instead

paulzuri, 2024-10-19:
## Bug Report

### What's the issue you encountered?

Console turning off instead of suspending when suspending from joycons, holding button HOME -> Suspend
No changes were made to atmosphere

### How can the issue be reproduced?

Turn on the console on emuMMC and try to suspend using the joycons, the console should turn off instead. Sometimes the issue also happens with the power button and sometimes it doesn't

### Crash Report

n/a

### System Firmware Version

19.0.0

### Environment?

Booting using hekate 6.2.2 official release

Additional Homebrew/Modules/Overlays:
- 90DNS Tester
- AIO Switch Updater
- Edizon + Overlay
- Fizeau + Overlay
- FTPD Pro
- Hekate Toolbox
- JKSV
- ldn_mitm + Overlay
- Linkalho
- MissionControl
- NX Activity Log
- NX FanControl
- Studious Pancake + Overlay
- SwitchTime
- sys-clk + Overlay
- Ultrahand

using emummc partition based

### Additional context?

n/a

borntohonk, 2024-10-19:
duplicate of https://github.com/Atmosphere-NX/Atmosphere/issues/2400

update all your sysmodules, and/or remove all the ones that lack updates.

paulzuri, 2024-10-19:
> duplicate of #2400
> 
> update all your sysmodules, and/or remove all the ones that lack updates.

any suggestions for tools in atmosphere or something to find the troubling module? i'm having a hard time finding it even after updating

borntohonk, 2024-10-19:
> > duplicate of #2400
> > update all your sysmodules, and/or remove all the ones that lack updates.
> 
> any suggestions for tools in atmosphere or something to find the troubling module? i'm having a hard time finding it even after updating

Just remove the entire /atmosphere/contents/ folder and your console will stop having your issue.

Introduce one by one folder until you figure out which it is.

closed 2024-10-20 by SciresM