Atmosphere-NX/Atmosphere #2406: Homebrew unable to start

sand-undertale, 2024-10-20:
### What's the issue you encountered?
When attempting to use any homebrew app it says "Unable to start software.", then says it's corrupted.

### How can the issue be reproduced?
Run homebrew through Title Takeover or Applet.

### Crash Report

N/A

### System Firmware Version

19.0.0

### Environment?

-Hekate with Atmosphere Pre-Release 18.0.0
-Attempted, failed.
-SysNAND

borntohonk, 2024-10-20:
> When attempting to use any homebrew app it says "Unable to start software.", then says it's corrupted.
Piracy is not supported.

sand-undertale, 2024-10-20:
as stated in the issue, it's homebrew, not piracy.

SciresM, 2024-10-20:
Piracy is not supported. This includes homebrew installed to the home menu using piracy patches.

closed 2024-10-20 by SciresM