Atmosphere-NX/Atmosphere #2407: Folders with `Archive` bit `on` appear as if corrupted (UMS mode)

idanmiara, 2024-10-20:
## Bug Report

Trying to access a folder that has the Archive bit set will be seen as if the data is corrupted, even though its intact.
Some storage devices keep the Archive bit set on folders, so if you copy the folders directly in UMS mode they are all as if corrupted - but once you unset the Archive bit on the copied files they are all fine.
The Archive bit is on by default for normal files, and off for folders. There is no issue with normal files with Archive bit on.
If you copy in MTP mode it seems that it doesn't set the Archive bit on the copied folders, so the issue is not present.

[image]

### How can the issue be reproduced?

1. Reboot into Hekate and connect the SD in UMS mode (tools -> USB tools -> USB Mass Storage -> SD Card)
2. Make 2 new folders: `a`, `b`.
3. Set the Archive bit on `a` (Windows11: right click on the folders, Properties, Advanced, tick `Folder is ready for archiving`)
4. Safely disconnect and reboot reboot atmosphere.
5. Try to access the folders with DBI (Browse SD Card) - `b` is OK. `a` is as if corrupted (as in attached photo)

### System Firmware Version

18.1.0

### Environment?
- Hekate 6.2.2
- Official Atmosphère 1.7.1
- EmuMMC partition-based

borntohonk, 2024-10-21:
> 5\. Try to access the folders with DBI (Browse SD Card) - `b` is OK. `a` is as if corrupted (as in attached photo)

Piracy is not supported.

idanmiara, 2024-10-21:
> > 5. Try to access the folders with DBI (Browse SD Card) - `b` is OK. `a` is as if corrupted (as in attached photo)
> 
> Piracy is not supported.

DBI has very nice file handling features, which are not related to piracy. It is hosted on Github a long time so it seems that it's fine by them too.

You may use your favorite File Explorer to verify they issue I reported, which is not piracy related.

borntohonk, 2024-10-21:
> DBI has very nice file handling features, which are not related to piracy. It is hosted on Github a long time so it seems that it's fine by them too.

The source code is not hosted on Github, your choice to word it as such is misleading. It's a russian-developed closed source tool for installed pirated titles.

Piracy is not supported.

SciresM, 2024-10-21:
Piracy is not supported.

SciresM, 2024-10-21:
Archive bit has special meaning to hos that is completely outside of atmosphere's control -- it marks a folder as being a file.

even besides the piracy commentary, this is straightforwardly not an atmosphere issue; any software interacting with the SD must be careful about archive bits, this is simply a requirement for horizon. 

macOS also has this problem.

closed 2024-10-21 by SciresM