Atmosphere-NX/Atmosphere #2410: Homebrew for Editing Atmosphere Settings via GUI on Switch

PoloNX, 2024-10-21:
## Feature Request

### What feature are you suggesting?
#### Overview:
- A homebrew that will allow users to modify the Atmosphere settings through a graphical interface. This homebrew enables modification of the Exosphere, Stratosphere, and all configurable .ini files.</br>

#### Smaller Details: 
- A library like [borealis ](https://github.com/xfangfang/borealis)can be used to create a good interface</br>

### What component do you feel this would best fit within?
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>

### Why would this feature be useful?
The user will be able to modify certain Atmosphere settings directly from their Switch.</br>

### Additionnal note
I can develop this homebrew and open a pull request. But if your against I don't want to create this tool for nothing so I prefer ask before starting the dev

wakoyname, 2024-10-21:
we would love to test it. start with the version 18.1.0 support atmos 1.7.1 for less bug

masagrator, 2024-10-22:
I was thinking about it few days ago, and I won't care if it's needed or not, I would use it almost every day. 
Remember that there are more settings than are included with template system_settings.ini, some of them are not even listed on switchbrew.

PoloNX, 2024-10-22:
Happy to ear that. Is it an obligation to use makefile directly ? Maybe I can use xmake and you can add this homebrew with a submodule.
Where can I find a list of all settings ?

masagrator, 2024-10-23:
I was thinking about making each entry with smaller height since by default imo they are too high + f.e. pressing + on highlighted setting to popup explanation what this setting does.

PoloNX, 2024-10-23:
I will try to improve the ui after doing something which is usable, but you're right

PoloNX, 2024-10-23:
@masagrator I implemented this quickly, what do you think?
[image]

SciresM, 2024-10-23:
I don't think this especially needs to be happening in the issue tracker for atmosphere.

closed 2024-10-23 by SciresM

PoloNX, 2024-10-23:
> I don't think this especially needs to be happening in the issue tracker for atmosphere.

I sent you a pm on discord

impeeza, 2024-10-23:
A suggestion from one which do not use discord:

the description could appears on the left side for each item you select on the right side.