Atmosphere-NX/Atmosphere #2411: Atmosphere issue black screen freeze after putting to sleep

Neptizler, 2024-10-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ updating Atmosphere ]</br>
[ updated from 1.7.1 ]</br>
[updated atmosphere/fusee/mission control/sigpatches also tried to reinstall tinfoil before i noticed the issue ]</br>

### How can the issue be reproduced?

[ * ]</br>
[After booting into atmosphere and putting the system to sleep I cannot wake from sleep. Almost seems to be in RCM mode and I have to do a hard power cycle to be able to use my jig and R4 card to access atmosphere ]</br>

### Crash Report
N/A

19.0.0</br>

### Environment?

- fusèe included with 1.8.0s, hekate v6.2.1 was Atmosphère 1.8.0s 
- Official release or unofficial build: 
  - [ Official build ]
 
- Do you have additional kips or sysmodules you're loading: mission control and sys-clk
- Homebrew software installed: [ mission control and sys-clk ]
- SysNAND:
 

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

drecklessf, 2024-10-22:
I am having the same issue and have not been able to figure out why

masagrator, 2024-10-22:
1. Duplicate
2. Not an atmosphere issue
3. Solution
https://github.com/Atmosphere-NX/Atmosphere/issues/2394#issuecomment-2420468826

closed 2024-10-24 by SciresM