Atmosphere-NX/Atmosphere #2412: Attempting To Open Cheat Menus on 1.8.0 Causes Atmosphere to Crash

foopatubc, 2024-10-22:
### Crash Report
From what I have tried, attempting to open cheat menus (Breeze using profile shortcut and EdiZon using Tesla Menu) on version 1.8.0 causes Atmosphere to crash, producing the following crash report:

https://gist.github.com/foopatubc/30273707eb411bde5cd1789d2cf97840

Aditionally, it seems that attempting to launch sysCFW with cheat files in the "contents" folder of Atmosphere also causes it to crash immediately after launch. From what I have seen, this renders cheats unusable on the newest version.

Thank you to the devlopers for all your hard work in creating this new version.

masagrator, 2024-10-22:
This is not an atmosphere issue, I bet. Breeze includes its own fork of dmnt, remove it and cheats won't crash system.

Ogurez123, 2024-10-22:
I had the same problem. Check if the hbl.nsp file exists in the atmosphere folder.
Maybe this will fix your problem.

foopatubc, 2024-10-24:
> This is not an atmosphere issue, I bet. Breeze includes its own set of sysmodules, remove them and cheats shouldn't crash system.

What makes me feel like it might be an issue with atmosphere is that the crashes occur not just due to breeze being launched but when sysCFW is launched while there are cheats in the content folder, even if they are not being used.

masagrator, 2024-10-24:
"Deleted some files"
If you don't know if correct ones, list what you deleted.

closed 2024-10-24 by SciresM