Atmosphere-NX/Atmosphere #2419: Not compatible with Arena of Valor

Bigguy0622, 2024-10-26:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

Fatal error occurs when entering the main menu of Arena of Valor. Sysclk installed

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]
- EmuMMC or SysNAND:
  - [ If using an EmuMMC, include whether it's partition-based or file-based. ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

masagrator, 2024-10-26:
Fill properly template.

borntohonk, 2024-10-26:
Since this game apparently is free on eshop ), i did the following:

1. Booted https://github.com/Atmosphere-NX/Atmosphere/releases/tag/1.8.0-prerelease on firmware version 19.0.0
2. Went to eshop, downloaded Arena of Valor (2.4GB)
3. Started the game
4. After intro accept sharing of nintendo account information to third-party
5. Accept EULA
6. Set user name
7. play tutorial

works fine for me, seems like just league of legends with a different name.

most likely a piracy issue // region block issue // dnsmitm issue // incognito or blanker issue

not atmosphere issue.

Bigguy0622, 2024-10-26:
Thank you! Does it matter if I apply sysclk on this game and would it cause atmosphere to crash?

borntohonk, 2024-10-27:
> Thank you! Does it matter if I apply sysclk on this game and would it cause atmosphere to crash?

That would not be an atmosphere issue, sys-clk has its own github page.

closed 2024-10-27 by SciresM